arduino/Arduino #3336: IDE using variable names instead of what they should be set to?

slango20, 2015-06-14:
it's a bit hard to explain, but it appears be using variable names instead of values, like so:
`Cannot run program "C:\Program Files (x86)\Arduino\{build.toolchain}{build.command.g++}" (in directory "."): CreateProcess error=2, The system cannot find the file specified`
(yes, I know it's a file not found, it's the underlying problem that causes that result)
the `{build.toolchain}` should be `hardware\tools\avr\bin\` and `{build.command.g++}` should be `avr-g++ [args here]`. the problem appears after manually editing the boards.txt to include the printrboard rev D (CDC bootloader or ISP, doesn't matter)

stickbreaker, 2015-06-15:
It looks like your boards.txt has an error.  Can you upload your boards.txt to github and post a link?

Chuck.

slango20, 2015-06-15:
the new board entry is this: 
```
printrbrd.name=Printrboard Rev D (CDC)
printrbrd.upload.tool=avrdude
printrbrd.upload.protocol=avr109
printrbrd.upload.maximum_size=130048
printrbrd.build.mcu=at90usb1286
printrbrd.build.core=at90usb1286
printrbrd.build.f_cpu=16000000L
printrbrd.upload.speed=115200
```
I copied this from a teensy boards.txt, although that boards.txt doesn't work at all (default boards work, just not teensy-defined boards)

it would also be nice to have some sort of clean fail mechanism in place for this, maybe have the program pop up an error dialog, or at least write *something* to log instead of being totally silent about the error

stickbreaker, 2015-06-15:
What are your:
build.board
build.variant
values?

Chuck.

slango20, 2015-06-15:
got it working now, I don't know why the teensy boards.txt omits those, or why I didn't add them from the stock boards, now I just have to do some board-specific tweaks.

closed 2015-06-15 by slango20

darrelhunt, 2016-06-21:
Sorry to revive this thread, but I can't find the information anywhere.  What did you do to get the Printrboard w/CDC Bootloader working with Arduino 1.6.6?