arduino/Arduino #3338: abs function incorrect return description

Netmisha, 2015-06-15:
http://www.arduino.cc/en/Reference/Abs

Returns
x: if x is greater than or equal to 0.
-x: if x is less than 0.

Should be:
Returns
x: if x is greater than or equal to 0.
x: if x is less than 0.

agdl, 2015-06-25:
@Netmisha no it is correct because if it is less than zero -X is a positive number and this is what you get. It is intended in this way

closed 2015-06-26 by agdl

bperrybap, 2016-10-03:
Actually, @Netmisha has a valild point. But the description on the reference page is correct, it is the macro that has the issue.
The macro in Arduino.h is this:
```
#define abs(x) ((x)>0?(x):-(x))
```

Technically it should be is:
```
#define abs(x) ((x)<0?-(x):(x))
```

The reason being is that there really a value of -0 and -0 is not the same value as 0
so the sign inversion should only be done for actual negative values.

In the bigger picture, why does Arduino still have these math function defines?
While this was a optimization trick back in the 80's or even 90's to avoid calling a function, gcc for quite some time has implemented these as an inline functions so there is no longer any value in using a macro for things like abs() and the macro comes with some pitfalls that can break some code.