arduino/Arduino #3339: Arduino Zero not appearing in boards manager for 1.6.5?

damellis, 2015-06-15:
With Arduino 1.6.4, I installed and used the SAMD (Arduino Zero) support just fine. When I ran 1.6.5, however, the Zero had disappeared from the boards menu. It looks like the package_index.json that 1.6.5 is downloading doesn't have the SAMD / Arduino Zero in it. This is on OS X.

ffissore, 2015-06-15:
Please remove the previous url from Preferences as it's not valid anymore. You'd also better clean up your ~/Library/Arduino15 folder from samd stuff. Boards manager shows SAMD core as expected. I'll also test this on mac. 
[image]

damellis, 2015-06-15:
This was after I removed my ~/Library/Arduino15 directory so I don't think it's a problem of having old preferences around.

damellis, 2015-06-15:
To clarify, after originally seeing the problem (after switching from 1.6.4 to 1.6.5), I quit 1.6.5, renamed my ~/Library/Arduino15 directory and then re-ran 1.6.5. The boards manager still didn't show the Arduino Zero.

ffissore, 2015-06-15:
Weird. Cache issue? Can you find samd core when opening http://downloads.arduino.cc/packages/package_index.json in your browser?

damellis, 2015-06-15:
Weird. Loading the URL in the browser, I saw the Arduino Zero. Then, when I
reran 1.6.5, Arduino Zero (SAMD) showed up in the boards manager. Not sure
what's going on.

cmaglie, 2015-06-15:
Are you using a proxy? this may explain the delay to propagate the cache

damellis, 2015-06-15:
I'm not using a proxy that I know of, although maybe MIT has something set
up?

cmaglie, 2015-06-15:
It may be, but I don't know, it would be nice if you discover and confirm that :-)

wholder, 2015-06-15:
I've tried removing the ~/Library/Arduino15 and let the 1.6.5 build recreate it, but still no Zero selection.  There is also nothing set in the URL in preferences.  If I open the Boards Manager, I can see an option for the Due under the SAM options, but I don't see the Zero listed anywhere.

Wayne

cmaglie, 2015-06-15:
@wholder the Zero is listed under SAM**D** that is different from SAM. You'll never find the Zero together with the Due, but you should scroll down a bit.

wholder, 2015-06-15:
Cristian, I don't see the Zero listed anywhere in the Boards Manager list.
Here's the complete set of things I see:

[image: Inline image 1]

Wayne

wholder, 2015-06-15:
Ok, this is weird...  I tried running the prior version IDE I had
installed, 1.6.4.  When I open the Boards Manager list, it showed the Zero
as an option to install.  Then, I quit 1.6.4 and tried 1.6.5 again.
However, 1.6.5 it still didn't show the Zero in the Boards Manager list.
So, I ran 1.6.4 again and let it install SAMD support for the Zero.  After
this, I can select the Zero in the 1.6.4 IDE, but it still does not appear
in the 1.6.5 install, or in the 1.6.5 install's Board Manager.

Wayne

wholder, 2015-06-15:
As a follow up to my last message, I removed the 1.6.5 install, downloaded
it again and then reinstalled it.  But, the Zero still does not show up as
a boar option to select, nor is SAMD support for the Zero listed in the
Boards Manager.  So, I can run using 1.6.4, but can't upgrade to the bug
fixes listed for 1.6.5.  Can anyone explain this?

Wayne

wholder, 2015-06-15:
Another follow up.  It gets even weirder.  Sometimes (mostly) when I start
up 1.6.4, the Zero does not show as a Board option I can select.  However,
if I open the Boards Manager, then close it (doing nothing), then the Zero
option shows up.  Still unable to see the Zero option in 1.6.5.

damellis, 2015-06-15:
It looks like the package_index.json file included in Arduino 1.6.5 (in Contents/Java/dist) doesn't include the Arduino Zero / SAMD. Is there some logic that's comparing the date of this file with the date of the latest package_index.json available online? Or another reason why the built-in file might get used over the one that's supposed to come from online? (And/or is it possible to add the Zero / SAMD to the included json file?)

wholder, 2015-06-15:
Now, suddenly, without me having made any changes, the Zero shows up in
both 1.6.4 and 1.6.5.  But, when I go to upload the blink example, I get:

Arduino: 1.6.5 (Mac OS X), Board: "Arduino Zero (Native USB Port)"

Cannot run program
"{runtime.tools.arm-none-eabi-gcc.path}/bin/arm-none-eabi-g++" (in
directory "."): error=2, No such file or directory

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

This code used to upload correctly earlier today and, as I said, I didn't
change anything.

Wayne

wholder, 2015-06-15:
I'm not sure if my posts on this are helping, but I was able to get the
compile step to work again by removing the SAM support for the "Due", then
reinstalling it.  Also, somewhere along the line, one of my two Zero boards
got itself into a state where it's no longer recognized by the IDE when I'm
using the Native port.  The Port entry under Tools no longer seemed to
recognize the other board as a Zero.  The Port menu showed only:

  /dev/cu.usbmodemfd131

without the "(Arduino Zero (Native USB Port))" string tacked on the end.  I
was able to fix this by switching to the Programmer port and uploading the
code that way.  Then, curiously, after that the Native port started to work
for uploads again.

One more thing, the RX and TX LEDs do not blink when code uploads, as was
done on the AVR-based boards.  Are these vestigial now that you're using
BOSSAC to handle the upload?

Wayne

ffissore, 2015-06-16:
Zero support is not listed in the bundled package index file because it was not yet released at the time of building the IDE. That *should* not be a problem to anyone since you need to go through boards manager to install zero core, hence updating that file.

Question is "why it doesn't get updated?". I double checked our code and no particular check is done. The only difference that may cause this issue is that IDE 1.6.5 will use your system proxy settings: maybe on macosx the "NO-OP" proxy does some sort of undesired caching?

ffissore, 2015-06-18:
I think we can close this. Thank you all for the heads up

closed 2015-06-18 by ffissore