arduino/Arduino #3340: 1.6.5: compilation fails when using the "byte" type in header files

mikecat, 2015-06-15:
I am using Arduino 1.6.5.

This code can be successfully compiled.

byte_test1.ino
```
struct hoge {
  byte a, b, c;
};

void setup() {
  hoge foo;
  foo.a = foo.b = foo.c = 123;
}

void loop() {

}
```

However, when I move ```struct hoge``` to a header file, I get some compile errors.
```byte_test2.ino``` and ```hoge.h``` are in the same sketch, same directory.

byte_test2.ino
```
#include "hoge.h"

void setup() {
  hoge foo;
  foo.a = foo.b = foo.c = 123;
}

void loop() {

}
```

hoge.h
```
#ifndef HOGE_H_GUARD_17F1CCC4_8F5B_4CAA_ADD2_3F5213947232
#define HOGE_H_GUARD_17F1CCC4_8F5B_4CAA_ADD2_3F5213947232

struct hoge {
  byte a, b, c;
};

#endif
```

the error message:
```
Arduino: 1.6.5 (Windows 7), ボード:"Arduino Uno"

In file included from byte_test2.ino:1:0:
hoge.h:5: error: 'byte' does not name a type
   byte a, b, c;
   ^
byte_test2.ino: In function 'void setup()':
byte_test2:5: error: 'struct hoge' has no member named 'a'
byte_test2:5: error: 'struct hoge' has no member named 'b'
byte_test2:5: error: 'struct hoge' has no member named 'c'
'byte' does not name a type

  This report would have more information with
  "コンパイル中の詳細な出力を表示する"
  ファイル > 設定 で有効にする
```

the more detailed error message (**** = my user name):
```
Arduino: 1.6.5 (Windows 7), ボード:"Arduino Uno"

ビルドオプションが変更されました。全体をリビルドしています。

C:\MyApps\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\MyApps\Arduino\hardware\arduino\avr\cores\arduino -IC:\MyApps\Arduino\hardware\arduino\avr\variants\standard C:\Users\****\AppData\Local\Temp\build261747110827034051.tmp\byte_test2.cpp -o C:\Users\****\AppData\Local\Temp\build261747110827034051.tmp\byte_test2.cpp.o 

In file included from byte_test2.ino:1:0:
C:\Users\****\AppData\Local\Temp\build261747110827034051.tmp\hoge.h:5:3: error: 'byte' does not name a type
   byte a, b, c;
   ^
byte_test2.ino: In function 'void setup()':
byte_test2.ino:5:7: error: 'struct hoge' has no member named 'a'
byte_test2.ino:5:15: error: 'struct hoge' has no member named 'b'
byte_test2.ino:5:23: error: 'struct hoge' has no member named 'c'
'struct hoge' has no member named 'a'
```

Why the "byte" type cannot be used in my header files?

Chris--A, 2015-06-15:
`byte` is included in Arduino.h,

It is automatically included into a sketch, however you must add it manually to your header.

ffissore, 2015-06-16:
Confirm

closed 2015-06-16 by ffissore