arduino/Arduino #3345: HardwareSerial::availableForWrite doing cli() without sei() for big buffers

jantje, 2015-06-16:
In Arduino IDE 1.6.5 I see the code below for HardwareSerial::availableForWrite
As you can see in case of (SERIAL_TX_BUFFER_SIZE>256) cli() is called.
I would expect it is needed to call sei() in the last part of the code.
However I do not see that.
Best regards
Jantje
```C++
int HardwareSerial::availableForWrite(void)
{
#if (SERIAL_TX_BUFFER_SIZE>256)
  uint8_t oldSREG = SREG;
  cli();
#endif
  tx_buffer_index_t head = _tx_buffer_head;
  tx_buffer_index_t tail = _tx_buffer_tail;
#if (SERIAL_TX_BUFFER_SIZE>256)
  SREG = oldSREG;
#endif
  if (head >= tail) return SERIAL_TX_BUFFER_SIZE - 1 - head + tail;
  return tail - head - 1;
}
```

cmaglie, 2015-06-16:
The interrupt are re-enabled with

```C++
#if (SERIAL_TX_BUFFER_SIZE>256)
  SREG = oldSREG;
#endif
```

we cannot use sei() because interrupts may be already disabled when entering the method, so we need to save the previous state in `oldSREG`, disable interrupts (even if they are already disabled) and, when done, restore the previous state.

closed 2015-06-16 by cmaglie

jantje, 2015-06-16:
txs
I didn't know that.

matthijskooijman, 2015-06-18:
I would consider using the [ATOMIC_BLOCK](http://www.nongnu.org/avr-libc/user-manual/group__util__atomic.html) macro for this, which makes code more readable and also allows disabling the atomicity at the top of the .cpp file, instead of having to check the buffer size in various places.

So, at the top of HardwareSerial.cpp you'd have:

    #if (SERIAL_TX_BUFFER_SIZE>256)
    #define TXBUF_ATOMIC_BLOCK ATOMIC_BLOCK
    #else
    #define TXBUF_ATOMIC_BLOCK(type)
    #endif

And for accessing variables:

    int HardwareSerial::availableForWrite(void)
    {
      TXBUF_ATOMIC_BLOCK(ATOMIC_RESTORESTATE) {
        tx_buffer_index_t head = _tx_buffer_head;
        tx_buffer_index_t tail = _tx_buffer_tail;
      }
      if (head >= tail) return SERIAL_TX_BUFFER_SIZE - 1 - head + tail;
      return tail - head - 1;
    }

(I didn't test the code, but you should get the idea. I'm not entirely
sure if defining TXBUF_ATOMIC_BLOCK(type) as empty is possible like this,
but it probably is)