arduino/Arduino #3346: Colors issue 1.6.5

pierrotm777, 2015-06-16:
The variable type boolean has no color for the words 'true' or 'false' !

Chris--A, 2015-06-16:
These are explicitly in the keywords.txt, I just guessed this was the colour they intended to use: 
https://github.com/arduino/Arduino/blob/master/build/shared/lib/keywords.txt#L57
https://github.com/arduino/Arduino/blob/master/build/shared/lib/keywords.txt#L73

ffissore, 2015-06-16:
I forgot it. Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-16 by ffissore

pierrotm777, 2015-06-16:
Colrs are specified but stay always black

ffissore, 2015-06-16:
Please re-download the hourly: you may have downloaded a yet-not-fixed version

pierrotm777, 2015-06-16:
It's ok now on 1.6.6 . What file i must to copy into the 1.6.5 IDE for solve this issue ?

ffissore, 2015-06-16:
`lib/theme/theme.txt` and pde.jar. Since we just started 1.6.6, it's safe for you to keep on using it: it's *almost* equal to 1.6.5

pierrotm777, 2015-06-16:
It's ok for me, thanks

otojunior, 2015-06-19:
I having this same problem. The property 'editor.fgcolor' in lib/theme/theme.txt doesn't working. The text it's always black
Where can I dowload version 1.6.6 ? Is the master branch? Have I to build it ?

Chris--A, 2015-06-21:
You can download it here:
http://www.arduino.cc/en/Main/Software

Scroll below the main download links and there is a section labelled 'Hourly Builds'

commit ffff480e (2015-06-16): Forgot to specify colour for LITERAL_BOOLEAN token type. Fixes #3346