arduino/Arduino #3349: Error highlighting wonky

Avamander, 2015-06-16:
[image]

I am using Ubuntu 15.04 and Arduino IDE release 2015/06/16 04:19.

This happens when IDE finds an syntax error and then takes the cursor there, when the cursor is moved with a mouse click a black bar (visible from screenshot) appears. This can be reproduced.

ffissore, 2015-06-17:
I can't reproduce it. The highlight bar is pinky from start to end

Chris--A, 2015-06-17:
No error on windows either.

pfeerick, 2015-06-21:
Not repeatable here. I unpacked a squeaky clean version of Arduino IDE v1.6.5 x32 onto Ubuntu 15.04 (from the package provided on the arduino site) and 'broke' the Blink sketch to get a similar error. I also get the pink highlight across the full line. 

[image]

closed 2015-06-22 by ffissore

Avamander, 2015-06-24:
Most likely some weird hardware+software and planetary state combination that causes this for me.