arduino/Arduino #3353: Board Manager - MessageDigest error?

pfeerick, 2015-06-17:
This is in relation to a 3rd party board package, so there is the possibility that this is due to an invalid board package file, but here goes...

Added 'http://boneskull.com/digistump-board-index/package_digistump_index.json' to the 'additional board manager URLs' list and tried to install the 'Digistump AVR Boards' by Christopher Hiller, and I get the following MessageDigest error at the bottom of the board manager (and in the log), and the java.lang.RuntimeException message is shown in the log. I have put in a request on the Digistump forum, and notice that someone else has had issues also recently. 

I am running v1.6.5 of Arduino, on Windows 7 x64, and the board package file is supposed to be 1.6.4+ compatible, the individual support files seem to download ok, and then the MessageDigest error rears it's ugly head...

Any help would be appreciated, and I happy to run any additional tests I can.

```
4e8c423fde901071259d679c3aaef97cf57c5229286057e40b73ffd4daffbf7a MessageDigest not available

java.lang.RuntimeException: java.security.NoSuchAlgorithmException: 4e8c423fde901071259d679c3aaef97cf57c5229286057e40b73ffd4daffbf7a MessageDigest not available
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:182)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.security.NoSuchAlgorithmException: 4e8c423fde901071259d679c3aaef97cf57c5229286057e40b73ffd4daffbf7a MessageDigest not available
	at sun.security.jca.GetInstance.getInstance(GetInstance.java:159)
	at java.security.Security.getImpl(Security.java:695)
	at java.security.MessageDigest.getInstance(MessageDigest.java:167)
	at cc.arduino.utils.FileHash.hash(FileHash.java:59)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:71)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:112)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:176)
	... 1 more
```

per1234, 2015-06-17:
It is an error in their board package file. They forgot to specify the checksum type for one of the Tools:
`"checksum": "4e8c423fde901071259d679c3aaef97cf57c5229286057e40b73ffd4daffbf7a"` should be `"checksum": "SHA-256:4e8c423fde901071259d679c3aaef97cf57c5229286057e40b73ffd4daffbf7a"`
I have submitted a pull request to them that fixes the problem: https://github.com/boneskull/digistump-board-index/pull/4

ffissore, 2015-06-17:
Closing as invalid, because upstream needs to handle this https://github.com/boneskull/digistump-board-index/pull/2

closed 2015-06-17 by ffissore

pfeerick, 2015-06-18:
Thanks for that, I thought someone would work out it quickly!