arduino/Arduino #3358: Board Manager: Unable to install Due on 1.6.5

TheForeignMan, 2015-06-18:
I have an Arduino Due I'd like to start using. I've been using Unos, Megas etc no problem. The newer IDEs have removed the Due support. I've been trying to install Due support through the Board Manager, but when I open the Boards Manager, I get : 

"Error downloading http://downloads.arduino.cc/packages/package_index.json"

and when I try to install Due support, I get :

"Error downloading http://downloads.arduino.cc/cores/sam-1.6.4.tar.bz2"

I understand I may be behind proxies, but in IDE version 1.6.5, that has been removed from the Preferences, so I can't change that. I have tried to allow the IDE past the Windows Firefall, but that has not changed anything.

I have manually downloaded "sam-1.6.4.tar.bz2" and extracted the contents to "C:\Arduino\arduino-1.6.5\hardware\arduino\sam" (I had to manually create the 'sam' folder). With this, I can see the options of Arduino Due Native USB Port and Arduino Due Programming Port when selecting the board, and when I look for it in the Ports, it can recognise the board at the port (eg "COM5 (Arduino Due (Programming Port))" ). When I try to compile anything, I get the error message :

Cannot run program "{runtime.tools.arm-none-eabi-gcc.path}\bin\arm-none-eabi-g++" (in directory "."): CreateProcess error=2, The system cannot find the file specified

Any other idea as to how I can install the Due? Is there an option for a manual installation?

ffissore, 2015-06-18:
The IDE is now using your system proxy settings. On windows, they are under Control Panel > Internet Options (on Windows 7 at least)

TheForeignMan, 2015-06-18:
Thanks for your answer, but the problem persists. 

Is there not a way of manually installing the Due support?

ffissore, 2015-06-18:
No. Can you confirm you're behind a proxy and you have set it to up correctly? Maybe your browser is using its own proxy settings, ignoring the system ones: that would explain why you can browse but the IDE fails. Have you also tried restarting the IDE once you've set up proxy settings?

TheForeignMan, 2015-06-18:
I might be under company proxies that can't be changed and are changed back to default when I change them. Yes, I have restarted the IDE after modifying the proxy settings. I have checked my browser's settings, and it is using the system settings.

ffissore, 2015-06-18:
With "might" you mean you don't know? Maybe your company proxy is also filtering traffic

TheForeignMan, 2015-06-18:
When I say "might", I mean I am under company proxy settings. I have just confirmed it.

ffissore, 2015-06-18:
As a workaround, you manually download the files specified in the dependent tools. Latest SAM core requirest tools `arm-none-eabi-gcc` version `4.8.3-2014q1` and `bossac` version `1.3a-arduino`. Download the `i686-mingw32` version of both and move the files into `%APPDATA%\Arduino15\staging`. It should work.
A more permanent solution is to contact your network administrators and make them open the proxy for the IDE

TheForeignMan, 2015-06-18:
I have downloaded both tools and moved them into the `%APPDATA%\Arduino15\staging` folder, so that it looks like this:
[image]
I have also left the files I manually downloaded for the SAM core in `C:\Arduino\arduino-1.6.5\hardware\arduino\sam`, so that it looks like this:
[image]
Unfortunately, I am still getting `Cannot run program "{runtime.tools.arm-none-eabi-gcc.path}\bin\arm-none-eabi-g++" (in directory "."): CreateProcess error=2, The system cannot find the file specified` in the IDE's command prompt when I try to compile a sketch for the Due. I can only assume that either I haven't copied the files over correctly, or the path to the `arm-none-eabi-g++` has not been defined somewhere.
Thanks for your help in this. If you can't think of any further potential solutions, I will go see the network admins. They usually take a week or so to come back, but I'll see what happens.

ffissore, 2015-06-18:
In staging folder there must no packages folder. Just **unpacked** zip/bz2 files files. Then open Boards Manager and install SAM core. It won't download anything because every file is already downloaded. It will just install them

ffissore, 2015-06-18:
Sorry I meant **packed** zip file

ffissore, 2015-06-18:
And I was mistaken again: correct folder is `%APPDATA%\Arduino15\staging\packages`, so, yes, a packages folder is needed

TheForeignMan, 2015-06-18:
Beautiful, this worked! Tante grazie per l'aiuto!

closed 2015-06-18 by ffissore

olivier57, 2015-08-19:
If this can help, I got the same message but this was because I forgot to unconnect Arduino wifi on my computer. Doing so and having an internet connection back, it worked well.

Olivier

dfc-cowmoo, 2015-08-21:
Hey I'm having the same problem and was wondering if anyone could help.

I've done the steps above but still receive the `Cannot run program "{runtime.tools.arm-none-eabi-gcc.path}\bin\arm-none-eabi-g++" (in directory "."): CreateProcess error=2, The system cannot find the file specified` message when compiling for the Due.

When using the boards manager to update, I receive the error `CRC doesn't match. File is corrupt'

When manually downloading the file `gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz` from http://arduino.cc/download.php?f=/gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz I still receive a corrupted file.

Any ideas?

Thanks for any tips.

Andy

adrian-nitu-92, 2015-10-15:
How does Arduino detect system settings on Linux? 
Which system settings? Because last I checked there wasn't any unified model for proxy system settings on Linux.

premkumars, 2016-07-26:
Open command prompt (windows start button, type cmd in search)
Click top left of command prompt, click properties, enable quick edit mode
Copy: setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true
Right click on command prompt, should paste

Source:http://stackoverflow.com/questions/16373906/address-family-not-supported-by-protocol-family-socketexception-on-a-specific

kb-, 2017-06-16:
I had the error too. Deleting the C:\Users\xxx\AppData\Local\Arduino15 folder worked for me.

justinclift, 2018-03-14:
For anyone else who needs to get an Arduino Due installed into the Arduino IDE while offline (eg on an isolated network), it is possible.  For a Win 7 PC here with Arduino IDE 1.8.5, a specific directory structure needed creating, with the files mentioned above being populated into it.

The files you'll need to download:

* http://downloads.arduino.cc/cores/sam-1.6.11.tar.bz2
* http://downloads.arduino.cc/bossac-1.6.1-arduino-mingw32.tar.gz
* http://downloads.arduino.cc/gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz

The `%APPDATA%` directory mentioned above (on this PC) resolves to `C:\Users\USERNAME\AppData\Local`.  So the directory structure needed was:

* `C:\Users\USERNAME\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\`
  * Extract files from sam-1.6.11.tar.bz2 into this.  Make sure the `cores`, `firmwares` (etc) structure is directly under the `1.6.11` folder after being extracted. (you may need to remove a `sam` subfolder that otherwise gets in the way)
* `C:\Users\USERNAME\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1`
  * Extract the gcc archive in here, such that the `arm-none-eabi`, `bin`, `lib`, and `share` folders are directly underneath it.  You'll probably need to remove the top level `gcc-arm-none-eabi-4.8.3-2014q1` from the archive structure to make this happen.

* `C:\Users\USERNAME\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino`
  * Extract `bossac.exe` from it's download directly into this folder

With this done, the Arduino Due shows up in the board list in the Arduino Due, sketches compile fine, etc.

As a data point, it's also possible to manually download [package_index.json](https://downloads.arduino.cc/packages/package_index.json) and [package_index.json.sig](https://downloads.arduino.cc/packages/package_index.json.sig), and place them in `C:\Users\USERNAME\AppData\Local\Arduino15`.  The "Boards Manager" dialog will then populate correctly and display the board list, even though being disconnected means it can't really do much.  Note the .sig file is needed, as the package_index.json is ignored without it.

anna-seppala, 2019-05-13:
How should I go about this on Linux?

justinclift, 2019-05-13:
Probably very similar, but the path to `Arduino15` would be somewhere in your home directory instead.

If you can figure out what that path is, you should be able to work out the corresponding paths from those above. :smile:

As a data point, I don't have any Arduino stuff around any more (doing different things), so probably not able to help in depth.  The above should be a good start though. :smile: