arduino/Arduino #3360: Windows: if username contains non ASCII char, IDE won't create correct Arduino15 folder

ffissore, 2015-06-18:
Because the output of `req query` is encoded someway
```

HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\Shell Folders
    AppData    REG_SZ    C:\Users\Ren\AppData\Roaming
```

bjorn-wiberg, 2015-06-21:
Same problem on a Swedish (CP850) Windows 8 installation (#3376).

ffissore, 2015-06-23:
Reopen since I'm not sure and need to test it

ffissore, 2015-06-23:
@bjorn-wiberg can you try downloading the hourly build? http://www.arduino.cc/en/Main/Software#hourly The problem should be fixed. Please start the IDE from command prompt using `arduino_debug.exe` so that any error message gets printed and you can paste them here

ffissore, 2015-06-23:
Positive feedback received from http://forum.arduino.cc/index.php?topic=329989.msg2288421#msg2288421. Closing

commit 98eb1a9e (2015-06-23): Re-introducing JNA, used only on Windows and only to access Windows Registry. Should fix #3360
closed 2015-06-23 by ffissore