arduino/Arduino #3362: java.lang.NoClassDefFoundError: processing/core/PApplet

ottojo, 2015-06-18:
Won't start

C:\Program Files (x86)\Arduino>arduino_debug.exe
Loading configuration...
Initializing packages...
Preparing boards...
Ungültige Bibliothek C:\Program Files (x86)\Arduino\libraries\Firmata in Librar
 can't use both 'src' and 'utility' folders. gefunden
Ungültige Bibliothek C:\Program Files (x86)\Arduino\libraries\Firmata in Librar
 can't use both 'src' and 'utility' folders. gefunden
Ungültige Bibliothek C:\Program Files (x86)\Arduino\libraries\Firmata in Librar
 can't use both 'src' and 'utility' folders. gefunden
Ungültige Bibliothek C:\Program Files (x86)\Arduino\libraries\Firmata in Librar
 can't use both 'src' and 'utility' folders. gefunden
Starting...
java.lang.NoClassDefFoundError: processing/core/PApplet
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
        at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:1
2)
        at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
        at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Class.java:348)
        at processing.app.Editor.addTools(Editor.java:926)
        at processing.app.Editor.buildToolsMenu(Editor.java:784)
        at processing.app.Editor.buildMenuBar(Editor.java:543)
        at processing.app.Editor.<init>(Editor.java:233)
        at processing.app.Base.handleOpen(Base.java:899)
        at processing.app.Base.handleOpen(Base.java:884)
        at processing.app.Base.handleOpen(Base.java:880)
        at processing.app.Base.handleNew(Base.java:768)
        at processing.app.Base.<init>(Base.java:477)
        at processing.app.Base.guardedMain(Base.java:226)
        at processing.app.Base.main(Base.java:137)
Caused by: java.lang.ClassNotFoundException: processing.core.PApplet
        at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 24 more

ffissore, 2015-06-18:
Have you downloaded version 1.6.5-r2?

ottojo, 2015-06-23:
yes, latest version.

ffissore, 2015-06-25:
Do you have a tool under C:\Program Files (x86)\Arduino\tools? Which tool is it?

ottojo, 2015-06-25:
there is a howto.txt and a folder "Mangler"

ffissore, 2015-06-26:
And do you have a tool in your sketchbook folder? Documents\Arduino\tools (if your sketchbook folder is the default one). Which tool is it?

ffissore, 2015-06-29:
anyway, you should remove it: it's the cause of your issue

ottojo, 2015-06-29:
yes i had tools in my documents folder. i deleted them and now everything is working again!

Thank You!

closed 2015-06-29 by ottojo