arduino/Arduino #3363: Compile with -fvtable-gc  for smaller binaries

yyyc514, 2015-06-18:
Ref: http://stackoverflow.com/questions/17433791/eliminate-unused-virtual-functions

Should this be a default?  I know a lot of graphics libraries subclass Print and then have to define write... but then you get things like write -> drawChar, fontTable, etc... and now you're linking in 2kb of stuff that a graphics (sprites, etc) only program will never use.  And the compiler happily links it all because `write` happens to be in your subclasses vtable and the compiler isn't trying hard enough to figure out if it's actually used or not.

Thoughts?

Chris--A, 2015-06-19:
From the latest build:

> avr-g++: warning: switch '-fvtable-gc' is no longer supported

yyyc514, 2015-06-19:
Oh:

http://stackoverflow.com/questions/17445907/replacement-for-fvtable-gc-in-gcc

Ok then, I guess that solves that.

closed 2015-06-19 by yyyc514