arduino/Arduino #3364: [Library Manager] Please Add Ucglib_Arduino

olikraus, 2015-06-18:
Please add Ucglib_Arduino to the Library Manager:

https://github.com/olikraus/Ucglib_Arduino

Ucglib is a graphics library for full color TFT and OLEDs. 
Ucglib is dicussed here: http://forum.arduino.cc/index.php?topic=214966.0

Thanks, Oliver

ffissore, 2015-06-19:
Added but your version is not semver complaint
```
2015/06/19 09:07:26 ... https://github.com/olikraus/Ucglib_Arduino.git

2015/06/19 09:07:45 ... ... tag 1.02pre3
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag ucglib_arduino_1.02pre3
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02

2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02-pre1
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02-pre2
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02-pre3
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02-pre4
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02pre5
2015/06/19 09:07:45 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:45 ... ... tag v1.02pre7
2015/06/19 09:07:46 Minor number must not contain leading zeroes "02"
2015/06/19 09:07:46 ... ... tag v1.02pre8
2015/06/19 09:07:46 Minor number must not contain leading zeroes "02"
```
Fix it, release another tag and it will become available

closed 2015-06-19 by ffissore

olikraus, 2015-06-19:
Thanks, I have fixed this.