arduino/Arduino #3365: 1.6.5 won't launch on Yosemite

ezjones, 2015-06-19:
I get this error on Console.log
6/18/15 8:47:53.590 PM com.apple.xpc.launchd[1]: (cc.arduino.Arduino.293972[1543]) Service exited with abnormal code: 255

1.5 works on the same machine. 

Is this Java related?

ffissore, 2015-06-19:
No, java is bundled with the IDE, you don't need to install it. Which version of macosx is?

ezjones, 2015-06-19:
10.10.3

ffissore, 2015-06-19:
Have you tried downloading it again? Can you share a screenshot?

ezjones, 2015-06-19:
it shows the splashscreen and then nothing happens. I get the error i shared before in Console.log

I downloaded 1.6.5 and several others.

Versions 1.5.x work and 1.6.x dont

tomasinouk, 2015-06-28:
Hi, I have the same issue. version **1.6.3** starts normally, but anything newer does not. 
Does not matter if it is nightly build or stable release. Tried (_1.6.4, 1.6.5, nightly build DL today_).

In regards the screenshot. It same as @ezjones mentioned. Only the splash screen, quits, when it shows `initialising packages`.
Running **OSX 10.10.3**

ffissore, 2015-08-05:
@tomasinouk do you get the same error message as @ezjones ? 
` 6/18/15 8:47:53.590 PM com.apple.xpc.launchd[1]: (cc.arduino.Arduino.293972[1543]) Service exited with abnormal code: 255` ?

jtuyen, 2015-08-07:
I'm getting the same symptoms and error in console as well using IDE 1.6.x.

2015-08-06 9:49:35.074 PM com.apple.xpc.launchd[1]: (cc.arduino.Arduino.5428[1287]) Service exited with abnormal code: 255

ffissore, 2015-08-07:
Have you tried 1.6.5-r3 which contain a mac specific fix?

jtuyen, 2015-08-07:
@ffissore yes, I have tried that version and the nightly build, same result.  I had to reinstall 1.5.x as it's currently working fine on my machine.

ffissore, 2015-08-07:
I'm out of options then. Any mac dev is welcome to give a hand

ezjones, 2015-08-08:
is there a way to launch it with more debug output?

ffissore, 2015-08-10:
Edit Info.plist, add a **JVMDebug** param so to make it end like so
```
<key>JVMArguments</key>
<array>
</array>
<key>JVMDebug</key>
<true/>
</dict>
</plist>
```
Run from command line

whoopy, 2015-08-14:
By process of elimination (tossing things out of my sketches folder until it launched) in my case it seems to be caused by TechWillSaveUs Gamer plugins in the **tools** sub folder, I had:

~/Documents/Arduino/tools
    |-- Animator/tool/Animator.jar
    |-- Painter/tool/Animator.jar

This is the crash I was getitng in the log. 1.6.5rc on Yosemite 10.10.4:
`<snip>`
2015-08-14 00:39:39.083 Arduino[79813:7527346] Arg 19: 'processing.app.Base'
java.lang.NoClassDefFoundError: processing/core/PApplet
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at processing.app.Editor.addTools(Editor.java:926)
	at processing.app.Editor.buildToolsMenu(Editor.java:784)
	at processing.app.Editor.buildMenuBar(Editor.java:543)
	at processing.app.Editor.<init>(Editor.java:233)
	at processing.app.Base.handleOpen(Base.java:899)
	at processing.app.Base.handleOpen(Base.java:884)
	at processing.app.Base.handleOpen(Base.java:880)
	at processing.app.Base.handleNew(Base.java:768)
	at processing.app.Base.<init>(Base.java:477)
	at processing.app.Base.guardedMain(Base.java:226)
	at processing.app.Base.main(Base.java:137)
Caused by: java.lang.ClassNotFoundException: processing.core.PApplet
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 24 more

ffissore, 2015-08-14:
Yep, remove them, they are not compatible with 1.6.x

ffissore, 2015-08-14:
BTW external tools are not related with the issue reported by @ezjones : his error looks as thrown by macosx launcher

whoopy, 2015-08-14:
OK - but if I go back to launching from finder rather than terminal I can still reproduce the original error in the console when these *incompatible* plugins are there..

com.apple.xpc.launchd[1]: (cc.arduino.Arduino.524864[986]) Service exited with abnormal code: 255

ffissore, 2015-08-14:
so you're saying this obscure error message is actually a consequence of having incompatible tools ?

whoopy, 2015-08-14:
Well if was for me

-  it seems to be worth trying to launch with a clean ~/Documents/Arduino folder first if you have problems

- When I enabled Debug I discovered from the warnings that I still had legacy stuff in my Sketches folder (old Adafruit hardware/atttiny/boards.txt files that seems to have been superseded by the new board management stuff) but that didn't turn out to be the case of the crash.
- I'd be interested to know if the crash log I got was a result of the extension being incompatible with the built-in jdk1.8.0_45.jdk or has the tools API changed ?

ffissore, 2015-08-14:
@ezjones @tomasinouk @jtuyen can you please verify you don't have custom tools in your SKETCHBOOK/tools folder?

@whoopy the IDE API changed about an year ago

ezjones, 2015-08-14:
I removed TechWillSaveUs Gamer tools and now it works!

ffissore, 2015-08-14:
:8ball:

tomasinouk, 2015-08-17:
@ffissore I removed whole folder `libraries` and removed whole folder `~/Documents/Arduino`
still the same error message. I will modify `Info.plist` for a debug output and see, if that would help.

Hopefully will be able to do it in few hours

ffissore, 2015-08-17:
@tomasinouk start it from a terminal: you'll have more chances to see some error message like those above

tomasinouk, 2015-08-19:
sorry for delay,

removed all _Arduino_ folders. (_had custom folder for sketches, in Dropbox folder, which have been used instead of default_) 
When started, the GUI noticed and notify that _sketch folder_ does not exist and switching to default. 
-> did not start. 

Run version **1.6.3**, which works. 

It started and used default sketch folder.

starting **1.6.5-r3** produces this

```sh
2015-08-19 15:34:57.328 Arduino[69194:1944817] Loading Application 'Arduino'
2015-08-19 15:34:57.329 Arduino[69194:1944817] Working Directory: '/Users/OSXuser'
2015-08-19 15:34:57.329 Arduino[69194:1944817] Java Runtime Path (relative): '/Users/OSXuser/Downloads/Arduino2.app/Contents/PlugIns/jdk1.8.0_45.jdk'
2015-08-19 15:34:57.329 Arduino[69194:1944817] Java Runtime Dylib Path: '/Users/OSXuser/Downloads/Arduino2.app/Contents/PlugIns/jdk1.8.0_45.jdk/Contents/Home/jre/lib/jli/libjli.dylib'
2015-08-19 15:34:57.330 Arduino[69194:1944817] Command line passed to application:
2015-08-19 15:34:57.330 Arduino[69194:1944817] Arg 0: '/Users/OSXuser/Downloads/Arduino2.app/Contents/MacOS/Arduino'
2015-08-19 15:34:57.330 Arduino[69194:1944817] Arg 1: '-Djava.class.path=/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/Classes:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/antlr.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/apple.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/arduino-core.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/bcpg-jdk15on-152.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/bcprov-jdk15on-152.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/commons-codec-1.7.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/commons-compress-1.8.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/commons-exec-1.1.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/commons-httpclient-3.1.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/commons-lang3-3.3.2.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/commons-logging-1.0.4.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/ecj.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/guava-18.0.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jackson-annotations-2.2.3.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jackson-core-2.2.3.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jackson-databind-2.2.3.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jackson-module-mrbean-2.2.3.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/java-semver-0.8.0.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jmdns-3.4.1.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jsch-0.1.50.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/jssc-2.8.0.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/pde.jar:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/rsyntaxtextarea-2.5.6.1+arduino.jar'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 2: '-Djava.library.path=/Users/OSXuser/Downloads/Arduino2.app/Contents/MacOS'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 3: '-DLibraryDirectory=/Users/OSXuser/Library'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 4: '-DDocumentsDirectory=/Users/OSXuser/Documents'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 5: '-DApplicationSupportDirectory=/Users/OSXuser/Library/Application Support'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 6: '-DCachesDirectory=/Users/OSXuser/Library/Caches'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 7: '-DSandboxEnabled=true'
2015-08-19 15:34:57.331 Arduino[69194:1944817] Arg 8: '-Dapple.awt.application.name=Arduino'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 9: '-Dapple.laf.useScreenMenuBar=true'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 10: '-Dcom.apple.macos.use-file-dialog-packages=true'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 11: '-Dcom.apple.smallTabs=true'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 12: '-Dcom.apple.macos.useScreenMenuBar=true'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 13: '-Xdock:name=Arduino'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 14: '-Dcom.apple.mrj.application.apple.menu.about.name=Arduino'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 15: '-Dfile.encoding=UTF-8'
2015-08-19 15:34:57.332 Arduino[69194:1944817] Arg 16: '-Xms128M'
2015-08-19 15:34:57.333 Arduino[69194:1944817] Arg 17: '-Xmx512M'
2015-08-19 15:34:57.333 Arduino[69194:1944817] Arg 18: '-splash:/Users/OSXuser/Downloads/Arduino2.app/Contents/Java/lib/splash.png'
2015-08-19 15:34:57.333 Arduino[69194:1944817] Arg 19: 'processing.app.Base'
java.lang.VerifyError: class org.bouncycastle.asn1.ASN1Primitive overrides final method equals.(Ljava/lang/Object;)Z
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
[... cut ...]
	at cc.arduino.contributions.GPGDetachedSignatureVerifier.verify(GPGDetachedSignatureVerifier.java:53)
	at cc.arduino.contributions.packages.ContributionsIndexer.isSigned(ContributionsIndexer.java:169)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:82)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:628)
	at processing.app.Base.<init>(Base.java:299)
	at processing.app.Base.guardedMain(Base.java:226)
	at processing.app.Base.main(Base.java:137)
```

hope that helps

tomasinouk, 2015-08-19:
after bit of googling:

```
ls /Library/Java/Extensions/ | grep bcprov
bcprov-jdk15-143.jar
```

```
ls /Users/OSXuser/Downloads/Arduino.app/Contents/Java/ | grep bcprov
bcprov-jdk15on-152.jar
```

There is a mismatch in versions. So I copy new version and backup old version

```
sudo cp /Users/OSXuser/Downloads/Arduino.app/Contents/Java/bcprov-jdk15on-152.jar /Library/Java/Extensions/
```

```
sudo mv /Library/Java/Extensions/bcprov-jdk15-143.jar /Library/Java/Extensions/back-up/
```

Now Arduino IDE starts.
tried with **1.6.5-r3** and with _nightly_ from **18/08/2015**

Not sure, if this brakes something else. But that was my problem. 
Thank you for your help @ffissore

ffissore, 2015-08-19:
Ah thank you. Indeed the cause of the error could only be a mismatch in that lib version. Your fix is correct: that was the source of the problem. I'll try to find a way to keep the JVM from loading system libs. Thank you!

ffissore, 2015-08-19:
FTR: that `sudo cp` is not needed. You only need to move `bcprov-jdk15-143.jar` out of `/Library/Java/Extensions`

tomasinouk, 2015-08-19:
I tried that at first `bcprov-jdk15-143.jar` -> `bcprov-jdk15-143.jar-bak` but that did not help in my case. Had to `cp` the right version.

ffissore, 2015-08-19:
0c2498a3672dc2e1836f6cea186103217436d39f contains a fix to this external jars loading issue. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly
The two flavours of this issue now have a fix and an explanation. Closing

closed 2015-08-19 by ffissore

dtbaker, 2016-02-09:
Hey just letting you know this error popped up again for me.

The fix was similar to what @tomasinouk mentioned ( copy the arduino version of `bcprov-jdk15on-152.jar` over to my local java ext directory )

````
dtbaker@laptop:~/software/arduino/arduino-1.6.7$ find|grep bcpr
./lib/bcprov-jdk15on-152.jar
dtbaker@laptop:~/software/arduino/arduino-1.6.7$ locate bcprov-jdk
/usr/java/packages/lib/ext/bcprov-jdk15-143.jar
dtbaker@laptop:~/software/arduino/arduino-1.6.7$ sudo cp ./lib/bcprov-jdk15on-152.jar /usr/java/packages/lib/ext/
```