arduino/Arduino #3370: Errors in openNextFile example sketch (and the corrections).

CosmickGold, 2015-06-19:
The example sketch for "openNextFile()", located at:
http://www.arduino.cc/en/Reference/FileOpenNextFile
has two flaws: (1) Once in a subdirectory, it never goes back to find more recent files in the root directory. (2) It then loops through "**nomorefiles**" forever.

Here is my corrected sketch, which I hope you can use to replace the flawed one:

#include <SD.h>

File root;
void setup()
{
  Serial.begin(9600);
  pinMode(10, OUTPUT);
  SD.begin(10);
  root = SD.open("/");
  printDirectory(root, 0);
}

void loop()
{
  // nothing happens after setup finishes.
}

void printDirectory(File dir, int numTabs)
{
  while(true)
  {
    File entry =  dir.openNextFile();
    if (! entry)
    {
      if (numTabs == 0)
        Serial.println("** Done **");
      return;
    }
    for (uint8_t i=0; i<numTabs; i++)
      Serial.print('\t');
    Serial.print(entry.name());
    if (entry.isDirectory())
    {
      Serial.println("/");
      printDirectory(entry, numTabs+1);
    }
    else
    {
      Serial.print("\t\t");
      Serial.println(entry.size(), DEC);
    }
  }
}

q2dg, 2015-06-19:
pinMode(10, OUTPUT); is necessary?

CosmickGold, 2015-06-19:
Yes.  I have read that pin 10 must be set as output -- even if not used -- to tell the processor to run it's SPI data pins (to the SD CARD) in Master Mode. The SD Card is at the end of the connection that runs in Slave Mode.

CosmickGold, 2015-06-20:
My project's sketch is large enough, that I could not properly run the above "printDirectory" function within it without running out of memory.  It seems the each time the line "File entry = dir.openNextFile();" exicuted, the new file was consuming additional memory, rather than replacing the previous file.

So here's a modified version of the above sketch that closes files before opening new ones.  Even within my larger sketch, this modified procedure works perfectly:

#include <SD.h>

File root;
void setup()
{
  Serial.begin(9600);
  pinMode(10, OUTPUT);
  SD.begin(10);
  root = SD.open("/");
  printDirectory(root, 0);
  delay(2000);
  Serial.println();
  Serial.println("Rewinding, and repeating below:" );
  Serial.println();
  delay(2000);
  root.rewindDirectory();
  printDirectory(root, 0);
  root.close();
}

void loop()
{
  // nothing happens after setup finishes.
}

void printDirectory(File dir, int numTabs)
{
  while(true)
  {
    File entry =  dir.openNextFile();
    if (! entry)
    {
      if (numTabs == 0)
        Serial.println("** Done **");
      return;
    }
    for (uint8_t i=0; i<numTabs; i++)
      Serial.print('\t');
    Serial.print(entry.name());
    if (entry.isDirectory())
    {
      Serial.println("/");
      printDirectory(entry, numTabs+1);
    }
    else
    {
      Serial.print("\t\t");
      Serial.println(entry.size(), DEC);
    }
    entry.close();
  }
}

agdl, 2015-07-02:
@CosmickGold it is onine now, if it is ok for you close the issue please

agdl, 2015-07-10:
I'm closing this. Reopen if needed.

closed 2015-07-10 by agdl