arduino/Arduino #3371: I can not find the platform.local.txt documentation.

jantje, 2015-06-19:
I just found out about the existence of platform.local.txt. I wanted to know how it worked in relation to arduino:samd (to make a debug version for zero) but I failed to find info about this.
I would expect to find the info here.
 https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification
I think something should be written about this.

PS as it seems I can not overwrite the compiler.c.flags in an platform.local.txt I will need a platform.txt anyway. I still think this should be documented.

matthijskooijman, 2015-06-20:
I added platform.local.txt but never got around to documenting it once the PR got merged.

The main usecase for platform.local.txt is to allow overriding values in a platform.txt file which is being tracked by e.g. git, so you prevent having modified files. I _think_ it does not allow modifying platform.txt from another directory (e.g. I don't think you can put a sketchbook/hardware/arduino/avr/platform.local.txt file to override arduino_install_dir/hardware/arduino/avr/platform.txt).

You say you cannot override the compiler.c.flags? I think I had an override for that for a while (to add C++11 support), though I'm not entirely sure now. Where is your platform.local.txt?

jantje, 2015-06-20:
>You say you cannot override the compiler.c.flags? I think I had an override for that for a while (to add C++11 support), though I'm not entirely sure now. Where is your platform.local.txt?

Well I tried only having a boards.txt and a platform.local.txt and that didn't work. So I can say that your thinking matches my experience.

I ended up modifying the boards.txt of the zero adding a debug board because it seems the -Og can not be just be anywhere on the command line.

per1234, 2018-02-19:
Closed by:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification/_compare/bee9ad447bf61393ded7c599a47b6672aac8d6e3...32a4de9d8df7bc1104861ffd561b7672dcc61d91
Anyone is welcome to improve on that.

I find that with Arduino IDE 1.8.5 or 1.9.0 beta build 36, Windows 10 64 bit, I am indeed able to override `compiler.c.flags` via platform.local.txt. If anyone is able to still reproduce that issue with the beta build then it would probably be best to open a new dedicated issue for it.

closed 2018-02-19 by per1234