arduino/Arduino #3373: 1.6.4. mising function / method in class not flagged

VaclavSal, 2015-06-20:
A declared class method NOT defined ( I hope I have the terminology correct) , AKA missing / not coded is NOT flagged. No error , no warning.
Compiles and runs just skipping over the code implementing the missing  method. 
ATmel Studio DOES post warning, without any specific option settings required, and also  compiles and runs the same way. 

Due to recent "be brief in your comments" suggestion resulting in no action , no further info will be provided.

Chris--A, 2015-06-20:
If the member is not used, no symbol is emitted for it by the compiler, and the linker should not produce a link error.

VaclavSal, 2015-06-20:
...AKA missing / not coded...

cmaglie, 2015-06-20:
If the method is not implemented (AKA missing / not coded) in the current compilation unit, it can be perfectly feasible that it could be implemented in another compilation unit (AKA another .cpp file), so the compiler can't emit an error, only the linker can (because it has visibility on **all** the compilation units).

What @Chris--A is trying to say is that if a method is not used (AKA is **never called**) the linker will not include it in the final build (to save precious flash memory space) so it won't search for it and will not complain if it's missing.

You can convince yourself with the following test:

```C++
class Test {
  public:
    void func();
};

Test t;

void setup() {
  // If you uncomment yhe following line:
  //t.func();

  // the link fails to with:
  // 11: undefined reference to `Test::func()'
}

void loop() {
}
```

If this doesn't answer your question, please provide a sketch (possibly as small as possible) that demonstrates the problem.

VaclavSal, 2015-06-20:
So why does ATmel Studio posts a warning about it?

cmaglie, 2015-06-20:
Can you show me a sketch that does display a warning in Atmel Studio and NOT in Arduino IDE? (possibly a small one..)

VaclavSal, 2015-06-20:
Nope, my sketch is well over few lines.
You just have to take my word for it, it DOES behave differently in IDE and VS.

matthijskooijman, 2015-06-20:
@VaclavSal can you then show the warning Atmel studio shows, then? I'm with @cmaglie on this one that it's expected behaviour to not show a warning or error. (If it's a warning - did you enable warnings in the preferences in the Arduino IDE?)

ffissore, 2015-06-22:
I think this question should be asked to Atmel Studio as it's the one that's behaving differently from expected.

cmaglie, 2015-06-25:
I've just tried the example in my previous post on Atmel Studio and it behaves exactly the same: no warnings displayed at all.

[image]

@VaclavSal please reopen the issue only when you have an example that shows the difference between the two IDE's.

closed 2015-06-25 by cmaglie