arduino/Arduino #3374: 1.6.4. addingh .zip libary

VaclavSal, 2015-06-20:
The .zip library is added to libraries via Add .zip library with following error message. 

Error compiling.
A library named FATS already exists

The library is not included ( no #include xxx ) when "include library" is used, only blank line(s) are added to .ino file.

matthijskooijman, 2015-06-21:
So this seems to tell you there is already a library with the same name. I guess you'll have to remove or rename that library before you can import this new one? What's the goal of this report? Do you think things are unclear, or do you have suggestions on how to better handle this? Do you think there is a bug?

VaclavSal, 2015-06-21:
Ok let me restate to clarify 
I am using "add zip library" menu - adding FATS library 
I receive 

Error compiling.
A library named FATS already exists

1. The compiler responded "compiling". 
    Compiling What?  Why? How? etc.  
    Take you best guess what needs to be done.

2. The compiler responded "A library named FATS already exists"
    The library does not existed before I used the menu. 
    My objective was to add the library, not to do QA to see what happens when the library already  exists  duh!  

YES THESE ARE BUGS. 

And this is an old bug also:

The library is not included ( no #include FATS in code  ) when "include library" is used, only blank line(s) are added to .ino file.

ffissore, 2015-06-22:
Is there a FATS folder in your sketchbook/libraries? Does it contain at least a `.h` file? If answers are, respectively, yes and no, then the behaviour is expected. If just push some changes that will prevent a library without header files from getting listed in the libs menu

closed 2015-06-22 by ffissore

VaclavSal, 2015-06-22:
YES

commit 5ebce32e (2015-06-22): Libraries: if missing header files, including a library doesn't generate empty lines. Fixes #3374
commit 6558afa2 (2015-06-22): Libraries: if missing header files, a lib doesn't even get listed. Fixes #3374