arduino/Arduino #3375: Does this really help?  just asking

VaclavSal, 2015-06-20:
During DEVELOPMENT adding #includes generates these. 
The compiler select  the #include file apparently as they appear in sequence in library directory. 
They are not errors, warnings etc. 
This can be hacked by moving them around or by adding #ifdef , but than it also requires moving any file such modified to a place where upgrading IDE will not erase them.
 
This post goes back to - who is checking the new releases to improve things and not just patching what has been broken for awhile. 

Actually if you really analyze these messages - minMsc is both used and unused ! 

Multiple libraries were found for "minMsc.h"
 Used: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minMsc
 Not used: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minMSC_Version_1.1
Multiple libraries were found for "minMsc_bulk.h"
 Used: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minMSC_Version_1.1
 Not used: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\minMsc
Multiple libraries were found for "LiquidCrystal_I2C.h"
 Used: C:\Documents and Settings\Vaclav\My Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Program Files\Arduino1.6.4._A\libraries\LCD_I2C

matthijskooijman, 2015-06-20:
What's your question exactly? Those messages indicate that you have multiple versions of a library installed (or more generally, multiple libraries providing the same .h file) and it indicates which library is being used. This is intended to help people understand what is going on and prevent confusion.

Not sure what your remark about checking releases means?

pfeerick, 2015-06-21:
I'm not sure on the purpose of this issue either, but I did notice that for that first include, minMsc is chosen, and for the second, minMsc_Version_1.1 is chosen (instead of minMsc again). The third case (LiquidCrystal_I2C.h) is understandable IIRC... the user copy is preferred to the copy in the the Arduino\libraries folder.. although that is also a non-standard (as in released-with-arduino) library?!

ffissore, 2015-06-22:
Yep, user liquidcrystal is preferred over bundled one, however having multiple minMsc libs in sketchbook/libraries brings to unexpected results. Usual workaround (which also become a code organization practice) is to switch sketchbook locations, so that each sketchbook contains a project or a collection of sketches that are somehow related

closed 2015-06-22 by ffissore

PaulStoffregen, 2015-06-22:
I'm the author of this warning message.

It was created because MANY users encountered problems when a previously installed library in their sketchbook folder would later conflict with Arduino when they upgraded to a new version of the IDE.  Typically those people had long forgotten they had placed a copy in their sketchbook libraries folder.  The SPI library was a very common case.  Before this special warning message, they would get only compile errors, regarding details of the code deep within the libraries.  Many forum threads showed users were unable to diagnose the problems.  Only very careful inspection of the pathnames would sometimes reveal the issue, and only when the conflict was with IDE-included library like SPI.

Since this warning message went into the IDE, those sorts of previously-common user questions have almost completely disappeared from the forums.  This warning message does indeed help a lot of people.

VaclavSal, 2015-06-22:
Paul,
thank for your reply. You did answer my question well and I appreciate you taking time for such detailed explanation.
Since I encountered this during development it did flag the issue as intended.
On very minor note - since it is intended as a warning it should be presented as such, but that is really splitting hair on my part.
Thank  you for all your contributions, you are the man.