arduino/Arduino #3376: 1.6.5-r2: International characters in Windows username: IDE won't launch

bjorn-wiberg, 2015-06-20:
In Arduino 1.6.5-r2 for Windows:

If one's Windows username contains international characters, such as ö (o with two dots / umlaut) in "Björn", the Arduino IDE shows the splash screen, upto "Preparing boards...", but then no IDE pops up. Instead the program exits.

Running arduino_debug.exe reveals that it chokes in the special character "ö", yielding an "Access denied" error:

C:\Program Files (x86)\Arduino>arduino_debug.exe
Loading configuration...
Initializing packages...
Preparing boards...
java.io.FileNotFoundException: C:\Users\Bj?rn\AppData\Roaming\Arduino15\preferen
ces.txt (Åtkomst nekad) [TRANSLATION: Access denied]
        at java.io.FileOutputStream.open0(Native Method)
        at java.io.FileOutputStream.open(FileOutputStream.java:270)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:213)
        at java.io.FileOutputStream.<init>(FileOutputStream.java:162)
        at processing.app.legacy.PApplet.createWriter(PApplet.java:574)
        at processing.app.PreferencesData.save(PreferencesData.java:115)
        at processing.app.Base.<init>(Base.java:343)
        at processing.app.Base.guardedMain(Base.java:226)
        at processing.app.Base.main(Base.java:137)
java.lang.RuntimeException: Couldn't create a writer for C:\Users\Bj?rn\AppData\
Roaming\Arduino15\preferences.txt
        at processing.app.legacy.PApplet.createWriter(PApplet.java:586)
        at processing.app.PreferencesData.save(PreferencesData.java:115)
        at processing.app.Base.<init>(Base.java:343)
        at processing.app.Base.guardedMain(Base.java:226)
        at processing.app.Base.main(Base.java:137)

C:\Program Files (x86)\Arduino>
--->8--- cut --->8---

This was not so in 1.6.4 at least.

If I run it as an administrator, it works. The reason for the "access denied" message, and the reason why it works as an administrator, is that it tries to create a directory alongside the real one in C:\Users:

---8<--- cut ---8<---
C:\Users>dir
 Volume in drive C is Packard Bell
 Volume Serial Number is 508B-D998

 Directory of C:\Users

2015-06-20  16:31    <DIR>          .
2015-06-20  16:31    <DIR>          ..
2015-04-19  15:08    <DIR>          Björn
2015-06-20  16:31    <DIR>          Bj"rn
2014-12-11  03:28    <DIR>          Public
               0 File(s)              0 bytes
               5 Dir(s)  205 368 213 504 bytes free

C:\Users>
--->8--- cut --->8---

Notice that the "ö" actually gets translated into the character ” (some typographic double quote?).

No ordinary (non-administrator) user has the right to create anything directly in C:\Users, hence it works as an administrator.

Hoping this can be fixed in some way (without changing username).

The most sensible approach would perhaps be if it could somehow get the short name version of the path in the %APPDATA% environment variable, so that...

APPDATA=C:\Users\Björn\AppData\Roaming

...would cause it to use:

APPDATA=C:\Users\BJRN~1\AppData\Roaming

...as the base for application data, and hence the preferences path would be:

C:\Users\BJRN~1\AppData\Roaming\Arduino15\preferences.txt

Best regards,
Björn

Chris--A, 2015-06-21:
This looks like a duplicate of the issue here: #3360

bjorn-wiberg, 2015-06-21:
Sorry, yes, my bad. This issue is a duplicate of  #3360.
Closing this issue.

closed 2015-06-21 by bjorn-wiberg