arduino/Arduino #3377: Bundled JRE doesn't work on all systems

matthijskooijman, 2015-06-20:
Apologies for the terseness of this report, but I'm just relaying a complaint from the #arduino IRC channel. Apparently the bundled JRE version only works on Fedora/Redhat systems. I haven't tried testing this on my Debian system yet.

	18:13 <_abc_> Okay, arduino-1.6.5 for linux was packaged with a
		      wrong java jre version and it won't work on non
		      RHEL/Fedora systems which do not provide glibc_2.4
	18:13 <_abc_> The fix is to edit the arduino shell script to use
		      the local java runtime engine which is not linked
		      against such crap
	18:13 <blathijs> _abc_: I was already wondering how long until
	              bundling java would cause problems
	18:13 <_abc_> glibc_2.4 was obsoleted in 2012
	18:14 <_abc_> and never existed on non RH non Fedora non Suse systems
	18:14 <blathijs> _abc_: Best course would be to report an issue
		      at the github repo (after checking recent reports
		      to see if it isn't already reported), care to do
		      that?
	18:14 <_abc_> Additionally, using LD_LIBRARY_PATH magic to try
	              to make it work, fails
	18:14 <_abc_> I will not bother. You do it.
	18:15 <blathijs> _abc_: In general, reports like these are
	              well-responded to.
	18:15 <_abc_> I do not have time for this, have other things to
	              do. Sorry.
	18:15 <_abc_> You do it, mention _abc_ on freenode.net found it,
	              that's all. Bye

matthijskooijman, 2015-06-20:
18:21 <_abc_> this is a Frankenstein machine I build things on. Mostly
	      Etch based (was). Most things are up to date, some not.
	      The 2.4 dep is a favorite on google, search for
	      'dependency broken glibc_2.4'18:21

ffissore, 2015-06-22:
Uhm, rudeness apart, I don't think we can fix or workaround this issue. A linux user unhappy with the bundled JRE can delete `java` folder inside IDE folder: the IDE will pick up system one. However _abc_ will still be disappointed since we switched to java 8 and I don't think there are Etch builds of openjdk-8. Since it's linux and it's twice free, upgrading is the way to go. Otherwise, stick with older versions.

ffissore, 2015-06-22:
BTW thank you for the report. IRC is missing from my feedback channels.

closed 2015-06-22 by ffissore

PaulStoffregen, 2015-06-22:
FWIW, 1.6.5 and the latest hourly seem to run fine on Ubuntu 14.04.

The hourly build appears to be compiled with a beta test version of the JDK, 1.8.0_60.  It seems to work fine, but perhaps using the latest stable 1.8.0_45 would be wiser?

ffissore, 2015-06-22:
Indeed. I'll reconfigure the CI server later tomorrow

ffissore, 2015-06-22:
Tracking this at #3396

matthijskooijman, 2015-06-22:
I just tried 1.6.5 on a Debian Jessie system, which runs without problems.
pick 3ec54b9 Support the "SEND_BREAK" CDC command

Checking the libc library shows that it provides the GLIBC_2.4 version normally:

        $ objdump -p /lib/x86_64-linux-gnu/libc.so.6 |grep GLIBC_2.4
        8 0x00 0x0d696914 GLIBC_2.4
                GLIBC_2.4 

This means that the original reporter's assertion that GLIBC_2.4 is only available on RHEL/Fedora is untrue, it seems more likely that older Linux versions simply don't provide it.

I'm still not so sure if providing a bundled JRE is a great idea, sinced it might create more problems than it solves, but apparently this is not one of those problems :-)