arduino/Arduino #3378: Inconsistent number of spaces in tab

per1234, 2015-06-21:
Using Windows 7 64 bit with Arduino IDE 1.6.5 Hourly Build 2015/06/19 06:42
Sometimes a tab at the end of the line is one space, other times it is two spaces.

Steps to replicate:
- File > New
- Press tab at the end of the line `void setup() {` - the tab is two spaces.
- Press tab at the end of the line `void loop() {` - the tab is one space.

I have the default tab values in preferences.txt:
```
editor.tabs.expand=true
editor.tabs.size=2
```

matthijskooijman, 2015-06-21:
Isn't the behaviour that tab aligns to the next multiple of 2 characters, so the number of spaces inserted depends on the length of the previous line? This is really how tabs originally worked as well, so I think this is not a bug?

per1234, 2015-06-21:
Yes I'm sure that's the cause now. That was not the behavior in previous versions of the IDE, for example in 1.6.3 the number of spaces in the tab is always the same and I prefer it that way. The change was probably caused by the new editor. It would be nice if there was an option in preferences.txt.

pfeerick, 2015-06-21:
Yes, it is probably due to the new editor. It now aligns all the tabs - so if you tab twice after ```void loop() {``` the cursor will be in vertical alignment with the single tab from ```void setup() {``` i.e. it's moving in multiples of two (when the tab size is set to 2) in relation to the start of the line, not just moving two characters to the right.

ffissore, 2015-06-22:
I've just checked http://www.jedit.org/ (previous editor was one of its subcomponents). Latest version of JEdit (5.1.0) behaves just like the new editor of the IDE. I think this was more a bug in the previous editor that got fixed upstream.
By default, tabs are already "soft" (emulated with spaces). If you prefer to stick to real tabs, close the IDE and change `editor.tabs.expand` to `false`

closed 2015-06-22 by ffissore

per1234, 2015-06-22:
Ok, I hadn't realized the cause of the inconsistency until matthijskooijman pointed out they are aligning.
>By default, tabs are already "soft" (emulated with spaces). If you prefer to stick to real tabs, close the IDE and change editor.tabs.expand to false

I prefer using spaces to tabs specifically because tabs usually do the alignment thing. Changing `editor.tabs.expand` to `false` does not cause a consistent tab length. I'll just get used to the new editor. Sorry for submitting a non-issue. Thanks everyone for taking the time to comment.