arduino/Arduino #3380: urgently needed: ANSI-C-like fprintf(), fscanf()

VogonJeltz, 2015-06-21:
urgently needed: ANSI-C-like fprintf(), fscanf()

Actually homebrewed fprintf_ works fine (SD file checked on PC), just fscanf_ muddles all up.

up to now used homebrewed function fscanf_  don't work for floats in larger files (>400 elements), neither by 1.6. nor by 1.6.3 nor by 1.6.5

used homebrewed functions:

Code:
int32_t  fscanf_ ( File * stream, const char fmtstr[], ... ) {
   const  int32_t   MAXSTRSIZE = 1024;
   char   str[MAXSTRSIZE];
   va_list   args;
   int32_t   i=0, cnt=0;
   int16_t   chr;

   va_start(args, fmtstr);

   strcpy(str, "");
   while ( (stream->available()) && (i < MAXSTRSIZE-1) ) {  
      chr = stream->read() ;
      if (chr>=0 && chr!='\n') {
           str[i]=(char)chr;    
           ++i;
      }
      else break;    
   }

   str[++i] = '\0';
                                                        
   cnt = vsscanf( str, fmtstr, args );
   va_end(args);

   return cnt;
}

//********************************************************************************​?***********

int32_t fprintf_ ( File * stream, const char fmtstr[], ... ) {
   char      str[1024];
   va_list   args;
   int32_t   num;
  
   va_start( args, fmtstr );
   num = vsnprintf(str, sizeof(str), fmtstr, args);
   stream->print(str);
   va_end( args );

   return num;
}

//********************************************************************************​***********

q2dg, 2015-06-21:
Maybe this thread is worthy: https://groups.google.com/a/arduino.cc/forum/#!topic/developers/7KpdLDgFsO0

VogonJeltz, 2015-06-21:
I don't understand unfortunately how your topic can read and write strings from an SD file and convert it to int or float. There also are no Arduino SD class functions available like, e.g., SD.freadFloat() or SD.freadInt16 or SD.freadInt32.