arduino/Arduino #3382: Arduino disconnect and reconnect causes exception when uploading if serial monitor left open

pfeerick, 2015-06-21:
Related partly to #3379 (actually discovered due to testing repeatability). 

An exception error will occur if the serial monitor is opened, the arduino is unplugged, plugged back in, and then uploaded to. When you click upload a second time, it disables the serial monitor properly and uploads.

```
java.io.IOException: jssc.SerialPortException: Port name - COM6; Method name - setEventsMask(); Exception type - Can't set mask.
	at processing.app.Serial.dispose(Serial.java:146)
	at processing.app.SerialMonitor.close(SerialMonitor.java:108)
	at processing.app.AbstractMonitor.suspend(AbstractMonitor.java:206)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2375)
	at java.lang.Thread.run(Thread.java:745)
Caused by: jssc.SerialPortException: Port name - COM6; Method name - setEventsMask(); Exception type - Can't set mask.
	at jssc.SerialPort.setEventsMask(SerialPort.java:279)
	at jssc.SerialPort.removeEventListener(SerialPort.java:1064)
	at jssc.SerialPort.closePort(SerialPort.java:1090)
	at processing.app.Serial.dispose(Serial.java:143)
	... 4 more
```

ffissore, 2015-06-23:
A fix to #3379 has just been pushed. Can you retry with the latest hourly build? http://www.arduino.cc/en/Main/Software#hourly

pfeerick, 2015-06-25:
Just tested the Arduino 1.6.6 2015/06/23 05:51 hourly build, the exception message is gone, but all that happens is the status area says 'Uploading to I/O Board...' - it doesn't compile and upload. If I click on upload a second time it does work.

ffissore, 2015-07-02:
Does avrdude say the com port in unavailable?

pfeerick, 2015-07-04:
I've just tried again with the latest hourly build, please ignore my previous message about it the message going away... with Hourly 1.6.6 2015/07/03 10:41 it is still there... I mustn't have waited long enought or held my tongue right before :-o error log reproduced below just in case it has moved....[which it has - AbstractMonitor and Editor are showing different line numbers]

Running the debug version of the IDE didn't give any more information. It looks like the gremlin is still in the IDE - which clears when clicking the compile button a second time.

```java.io.IOException: jssc.SerialPortException: Port name - COM6; Method name - setEventsMask(); Exception type - Can't set mask.
	at processing.app.Serial.dispose(Serial.java:140)
	at processing.app.SerialMonitor.close(SerialMonitor.java:108)
	at processing.app.AbstractMonitor.suspend(AbstractMonitor.java:119)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2423)
	at java.lang.Thread.run(Thread.java:745)
Caused by: jssc.SerialPortException: Port name - COM6; Method name - setEventsMask(); Exception type - Can't set mask.
	at jssc.SerialPort.setEventsMask(SerialPort.java:279)
	at jssc.SerialPort.removeEventListener(SerialPort.java:1064)
	at jssc.SerialPort.closePort(SerialPort.java:1090)
	at processing.app.Serial.dispose(Serial.java:137)
	... 4 more
```

bcraigie, 2016-11-26:
The problem still occurs in 1.6.9 on Windows 10.  It seems to alternate between Can't Set Mask and Port Not Found.

oqibidipo, 2016-11-28:
Does it occur in 1.6.13?

leaninonlife, 2017-04-30:
I am experiencing the same error on an Arduino Leonardo Clone w/built in CC3000 Wifi (DFRobot WiDo) Board. I'm trying to determine if it is related to the issue I'm experiencing where the board will run for a few seconds after it I flash it, then halts in its tracks with a red light steady and pin 13 stuck in on position. It will report an attempt to initialize the Blynk library and communicate via Wifi, but stops before it can ever actually establish a network connection.

axellettieri, 2018-01-07:
Same for me, after an update of windows 10.
IDE 1.8.5, on Arduino Due.

rugo100, 2018-02-23:
is there any fix for this issue? (IDE 1.8.5 on ESP8266)

PCees4u, 2018-03-16:
Exactly same proplem! (IDE 1.8.5 on ESP-WROOM-32)

BSFishy, 2018-09-27:
Same here. (IDE 1.8.7 on Adruino Uno w/ ESP8266-01 connected)

simitenbiri, 2019-01-08:
Thanks pfeerick , I had a similar issue and your comment helped

SergiuToporjinschi, 2019-02-06:
same issue here arduino 1.8.8 on esp8266-01

Lopes2700, 2019-03-14:
same issue in Arduino 1.8.8 on esp32

Buzzardbait, 2019-07-11:
Same issue with Tongze ESP8266/FTD232R Bridge and Ardurino IDE 1.8.9
Sensitive to Serial Monitor being on when uploading?

Buzzardbait, 2019-07-12:
Oh, that was silly (and embarrassing)! The FTD FT232R looked like it could power the Tongze shield when not mounted on the UNO/MEGA but it eventually overheated and symptoms were as described above. Ended up powering shield by mounting on the MEGA board and being careful to only connect Tx, Rx and Gnd (not 3.3v) from FT232R to the serial debug port. Shield works fine now (but probably not via MEGA D0 and D1 pins). Need to work on that.

deviprasannaraju, 2020-04-10:
java.io.IOException: jssc.SerialPortException: Port name - COM3; Method name - setEventsMask(); Exception type - Can't set mask.
	at processing.app.Serial.dispose(Serial.java:166)
	at processing.app.SerialMonitor.close(SerialMonitor.java:147)
	at processing.app.AbstractMonitor.suspend(AbstractMonitor.java:115)
	at processing.app.Editor$UploadHandler.run(Editor.java:2069)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - COM3; Method name - setEventsMask(); Exception type - Can't set mask.
	at jssc.SerialPort.setEventsMask(SerialPort.java:279)
	at jssc.SerialPort.removeEventListener(SerialPort.java:1064)
	at jssc.SerialPort.closePort(SerialPort.java:1090)
	at processing.app.Serial.dispose(Serial.java:163)
	... 4 more

Jeevesh8, 2020-09-20:
@deviprasannaraju did you solve ?

Buzzardbait, 2020-09-20:
That was a long time ago and I haven't had any problems with the MEGA or UNO boards in a long time.

ingredyanhaia, 2020-09-28:
Arduino Uno [v1.8.13][win10]
Mesmo problema aqui, não consegui seguir ainda.

[image]

Makcal, 2021-09-28:
The problem is still relevant

TomMandle, 2022-01-04:
The problem is still there and relevant.  Is there a fix for it?  Thank you.

ireluq, 2022-03-24:
Hi, did someone find the solution to this problem? 

Still relevant (1.8.16) and looking desperately for a solution.

bwanaaa, 2022-04-18:
same here. using arduino IDE 1.8.19 and arduino pro micro.

per1234, 2022-12-01:
Hi all. Please try with the latest version of Arduino IDE 2.x and comment here to let us know whether the problem still occurs.

The Arduino IDE 2.x download links are listed here:

https://www.arduino.cc/en/software/