arduino/Arduino #3383: Gto to the line error on click error

pierrotm777, 2015-06-21:
I should to know if it's possible to have a cool feature , go automatically to the line error when i click on the debug error message 
[image]
!

ffissore, 2015-06-22:
Indeed cool. Can't promise, but I'll see what I can do

pierrotm777, 2015-06-22:
Cool, thanks :-)

ffissore, 2015-09-25:
Also see #976

lmihalkovic, 2018-05-01:
this is all the code that is missing to do it

```
  private class ErrorDetector extends MouseAdapter {
    
    //sketch_dec04a:7:1: error: 'Cls' was not declared in this scope
    Pattern perr = Pattern.compile("^(\\w+):(\\d+):(\\d+):\\W+(\\w+):");
    Editor editor;
    
    ErrorDetector(Editor editor) {
      this.editor = editor;
    }

    @Override
    public void mouseClicked(MouseEvent event) {
      JTextPane textArea = EditorConsole.this.consoleTextPane;
      int offset = textArea.viewToModel( event.getPoint() );
      try {
        int start = Utilities.getRowStart(textArea, offset);
        int end = Utilities.getRowEnd(textArea, offset);
        String text = textArea.getText(start, end - start);
        Matcher m = perr.matcher(text);
        if (m.find()) {
          String file = m.group(1);
          String line = m.group(2);
          String col = m.group(3);
          SketchLocation loc = new SketchLocation(file, line, col);
          editor.gotoLocation(loc);
        }
        
      } catch (BadLocationException e) {
        e.printStackTrace();
      }
    }

  }  
```
It could be fancier to change the cursor in the console when the cursor is over an error, letting users know that they can click... but right now clicking works fine with this code.

lmihalkovic, 2018-05-01:
ok... so with some changes, I did the highlighting of the error ref... the color is hardcoded for now. but I think it is useful... should now also be simple to implement something like

- CTRL+ALT+[  --> jump to previous error
- CTRL+ALT+]  --> jump to next error
- ensure that the console scrolls back to the position of the first error it contains if there is one (otherwise stay wherever)

[image]

lmihalkovic, 2018-05-01:
... and then it becomes possible to __click-to-jump__ to a Warning location. 

[image]