arduino/Arduino #3385: arduino IDE hangs on Compile instead of reporting syntax errors

pkg40, 2015-06-21:
I see this happening very frequently with a variety of syntax errors. The latest example is that the following code causes IDE to hang inside of class constructor:

 if (stepType==STEPTYPESIXTEENTH) {
    digitalWrite(msPin[2],HIGH);
    digitalWrite(msPin[1],HIGH);
    digitalWrite(msPin[1],HIGH);
  }
  else if (stepType==STEPTYPEEIGHTH) {
    digitalWrite(msPin[3],LOW);
    digitalWrite(msPin[2]=HIGH);
    digitalWrite(msPin[1]=HIGH);
  }
  else if (stepType==STEPTYPEQUARTER) {
    digitalWrite(msPin[3]=LOW);
    digitalWrite(msPin[2]=HIGH);
    digitalWrite(msPin[1]=LOW);
  }
  else if (stepType==STEPTYPEHALF) {
    digitalWrite(msPin[3]=LOW);
    digitalWrite(msPin[2]=LOW);
    digitalWrite(msPin[1]=HIGH);
  }
  else if (stepType==STEPTYPEFULL) {
    digitalWrite(msPin[3]=LOW);
    digitalWrite(msPin[2]=LOW);
    digitalWrite(msPin[1]=LOW);
  }
  else {
    digitalWrite(msPin[3]=LOW);
    digitalWrite(msPin[2]=LOW);
    digitalWrite(msPin[1]=LOW);
  }

 Note that there are 13 instances of the digitalWrite function with incorrect number of operands. Any fewer than 13 and the compiler correctly spits out errors. 13 or more and the compiler hangs with following in the status window:

Using library SPI in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library Wire in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library Adafruit GFX Library in folder: C:\Users\pkg\Google Drive\Exclude_Personal\Arduino\libraries\Adafruit-GFX-Library 
Using library Adafruit SSD1306 in folder: C:\Users\pkg\Google Drive\Exclude_Personal\Arduino\libraries\Adafruit_SSD1306 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\eightanaloginputs -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Users\pkg\Google Drive\Exclude_Personal\Arduino\libraries\Adafruit-GFX-Library -IC:\Users\pkg\Google Drive\Exclude_Personal\Arduino\libraries\Adafruit_SSD1306 C:\Users\pkg\AppData\Local\Temp\build6940767360242408296.tmp\sketch_jun15a.cpp -o C:\Users\pkg\AppData\Local\Temp\build6940767360242408296.tmp\sketch_jun15a.cpp.o

pkg40, 2015-06-21:
Further details:

IDE version 1.6.4.  (downloaded from arduino.cc.)
Windows 8.1

per1234, 2015-06-21:
That's a problem with 1.6.4, just install 1.6.5 and it will be fixed

ffissore, 2015-06-22:
I confirm. It's a bug we fixed with 1.6.5

closed 2015-06-22 by ffissore