arduino/Arduino #3387: Double clicking after a open curly braces should select the block

ffissore, 2015-06-21:
This happened with the previous editor and it's not happening anymore with the new one. Try to reimplement it and report back at http://forum.arduino.cc/index.php?topic=330772

ffissore, 2015-06-22:
Also see upstream https://github.com/bobbylight/RSyntaxTextArea/issues/90

artag, 2015-08-22:
There's some sort of animation happening in parentheses pairs on 1.6.5 - for every character I type while the cursor is between the parentheses, the highlight  on one end blinks. I can tolerate that when I move the cursor in, but not on every typed character. 

If this is intentional, can I turn it off please ?

Same goes for every automated function  or syntax colouring. I absolutely appreciate some people love them : for the rest of us, please give a way to turn it off.

And github, please get rid of that popup that happens when I type a colon. Or a commercial-at. Just let the keys do what the keytops say.

Chris--A, 2015-08-23:
I agree, the animation is quite distracting and should be prevented, or only show the final result (no animation while typing). I could see a quick fix as adding a delay before it it shown (hidden straight away on type) then it should effectively be invisible until the last character is typed (if the user types above a certain minimum speed). If I can find the particular section in RSTextArea, I'll give it a try.

I think your're underestimating the people "for highlighting" as a minority. Syntax highlighting is an invaluable feature. For sure, I think the colors are too flat, there is not enough contrast between key aspects like comments which should stand out, not blend into the white background. Regardless however, un-highlighted syntax is terrible when working on large/complicated sources.

ffissore, 2015-08-24:
@artag your comment must go into a separate issue: it's off topic with this one. Also, if you really want github to hear you, write them, not on Arduino repo :)

nickgammon, 2015-10-18:
@ffissore - I agree it is off-topic, however it appears to be referring to the Arduino IDE. 

http://forum.arduino.cc/index.php?topic=353730

I'll open a new issue for it.

per1234, 2017-07-01:
It turns out that https://github.com/bobbylight/RSyntaxTextArea/issues/90 is unrelated to this issue. In that issue "Block Selection" has a completely different meaning (see http://fifesoft.com/blog/?p=291). To be more clear, this is how "Block Selection" looks in Notepad++ (where it is known as "Column Mode Select"):

--------------
[image]

--------------
versus standard select:

[image]

per1234, 2023-07-27:
Closing as resolved because the equivalent capability is available in Arduino IDE 2.x via the "**Select to Bracket**" command.

There is no keyboard shortcut associate with the command by default (meaning it is only available from the ["Command Palette"](https://code.visualstudio.com/docs/getstarted/userinterface#_command-palette) menu), but you can assign your own custom shortcut:

1. Select **File > Advanced > Keyboard Shortcuts** from the Arduino IDE menus.
   The "**Keyboard Shortcuts**" tab will open in the Arduino IDE window.
1. Type `Select to Bracket` in the "**Search keybindings**" field of the "**Keyboard Shortcuts**" tab.
1. Hover the mouse pointer over the "**Select to Bracket**" row in the search results.
   A pencil icon ("Edit Keybinding") will appear at the left side of the row.
1. Click the pencil icon.
   The "**Edit Keybinding for \_\_\_**" dialog will open.
1. The input field in the dialog contains a text representation of the keyboard shortcut. Adjust this according to your preferences.
   **ⓘ** Notes about the syntax:
   - For most keys you can simply use the name of the key (i.e., `a`, `3`, `/`, `f2`).
   - Special key names:
     - `cmd` - <kbd>**Command**</kbd> key (<kbd>**⌘**</kbd>) on macOS.
     - `ctrlcmd` - <kbd>**Ctrl**</kbd> key on Linux/Windows, <kbd>**Command**</kbd> key (<kbd>**⌘**</kbd>) on macOS.
   - Use the `+` separator for key combinations (e.g., `ctrl+s`).
   - Use a space separator for [key sequences](https://github.com/eclipse-theia/theia/tree/master/packages/keymaps#key-sequences) (AKA "[chords](https://code.visualstudio.com/docs/getstarted/keybindings#:~:text=current%20context.-,Chords,-%28two%20separate%20keypress)") (e.g., `ctrl+k ctrl+i`).
1. Click the "**OK**" button in the dialog.
1. Click the **X** icon on the "**Keyboard Shortcuts**" tab.

closed 2023-07-27 by per1234