arduino/Arduino #3388: Executing IDE loads ALL previously opened projects

VaclavSal, 2015-06-21:
Please confirm it is a feature and not a bug.

Automatically loading previously opened SINGLE project is usually an option and such feature should be documented somewhere. 

Suspected multiple mouse clicks but observed this behavior when Windows Startup includes Arduino IDE

Chris--A, 2015-06-22:
This is confirmed. It has happened through many versions of the IDE. I just run into it so little that I never think about it.

per1234, 2015-06-22:
It happens with File > Quit but not with File > Close or the standard close window X button. I'm pretty sure this is the intended behavior of the IDE and not a bug.

VaclavSal, 2015-06-22:
So the real fix would be to  use "Tool tip" or F1 or text in status bar  to give user some clue what it actually does in each instance, 
I think issues like this need to be addressed so developers can work on real problems  and not be bothered with undocumented "features" possibly perceived as bugs.

closed 2015-06-22 by ffissore