arduino/Arduino #3391: Environment documentation updates

per1234, 2015-06-22:
Some suggestions for changes  to www.arduino.cc/en/Guide/Environment

Menu item descriptions section:

File
- Quit

  Closes all IDE windows. The sketches will be automatically reopened the next time you start the IDE.

Sketch
- Upload Using Programmer

  This will overwrite the bootloader and so you will need to use Tools > Burn Bootloader to be able to Upload to USB serial port. Allows you to use the full capacity of the Flash memory for your sketch.  

- Change **Import Library** to **Include Library**

============

[Third-Party Hardware](www.arduino.cc/en/Guide/Environment#thirdpartyhardware) section:

change:
>For details on creating packages for third-party hardware, see the platforms page on the Arduino Google Code developers site. 

to:

For details on creating packages for third-party hardware, see the [Arduino IDE 1.5 3rd party Hardware specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification).

agdl, 2015-06-25:
@per1234 done if it is good for you please close the issue :)

per1234, 2015-06-25:
I don't see the File > Quit documentation. Quit has a non-obvious behavior as demonstrated by https://github.com/arduino/Arduino/issues/3388. Another thing I just noticed is in the [Sketchbook section](http://www.arduino.cc/en/Guide/Environment#sketchbook):
```
'''Beginning with version 1.0, files are saved with a .ino file extension. Previous versions use the .pde extension. You may still open .pde named files in version 1.0 and later, the software will automatically rename the extension to .ino. 
```
I think the three apostrophes at the start of the paragraph are typos.

Of course these are just suggestions, feel free to implement them or not. Thanks!

agdl, 2015-06-25:
Take a look now please

closed 2015-06-25 by per1234