arduino/Arduino #3392: Hyperlinks with ampersand in comments are truncated

per1234, 2015-06-22:
- add the text `// https://forums.adafruit.com/viewtopic.php?f=25&t=12170` to a sketch
- click on the URL
- the browser opens https://forums.adafruit.com/viewtopic.php?f=25 instead of https://forums.adafruit.com/viewtopic.php?f=25&t=12170

ffissore, 2015-06-22:
Doesn't happen on linux, so I think it's because there are no double quotes around the url [here](https://github.com/ffissore/Arduino/blob/master/arduino-core/src/processing/app/windows/Platform.java#L137-137). I'll check it as soon as I can get my hand a windows box

Chris--A, 2015-06-22:
I'll check it now.

Chris--A, 2015-06-22:
Its quite strange this one. I couldn't get it working with quotes. It just opened up a console named with the URL.

Directly using the console is no better.
This only searches for *cheese*:
`start https://google.com?q=cheese&q=bread`

This opens up a console:
`start "https://google.com?q=cheese&q=bread"`

To get it working, an empty set of quotes is required before the quoted URL.
Now a browser opens and searches for *bread*:
`start "" "https://google.com?q=cheese&q=bread`

Here is the commit I ended up with: https://github.com/Chris--A/Arduino/commit/cac568f44c3dd007d29d24c9aedd6cc86804fffc

ffissore, 2015-06-22:
Thanks @Chris--A , I'll test it on my windows box later tomorrow. According to this, your empty double quotes is assigned to the `title` param, while the url to the `filename`: maybe `start` opens the url specified with title if not filename is given, but `&` are forbidden in titles and get removed

ffissore, 2015-06-22:
"this" was this https://www.microsoft.com/resources/documentation/windows/xp/all/proddocs/en-us/start.mspx?mfr=true

ffissore, 2015-06-23:
@Chris--A looks good. Mind making a plain PR? So we keep on dancing the usual dance

Chris--A, 2015-06-24:
No worries, will push it through

PR #3410 (2015-06-25): Fix missing trailing parameters when opening URL's﻿ (WIN)
This is a PR for the issue #3392. On Windows platforms query string parameters separated using `&` are removed due to OS behavior. This PR wraps the URL in quotes allowing the URL to be passed unmodified to the browser.

Closes #3392
closed 2015-06-25 by ffissore