arduino/Arduino #3394: Compiling with a 3rd party board selected causes java.io.FileNotFoundException error

per1234, 2015-06-22:
Windows 7 64 bit Arduino IDE 1.6.6 Hourly Build 2015/06/19 06:42
With 3rd party board installed to sketchbook/hardware folder or installed via Boards Manager selected I get the following error when I compile any sketch:
```
java.io.FileNotFoundException: C:\Program Files (x86)\arduino-nightly\hardware\arduino\avr\bootloaders\ariadne\ariadne_atmega2560.hex (The system cannot find the path specified)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at java.io.FileReader.<init>(FileReader.java:72)
	at processing.app.helpers.FileUtils.readFileToListOfStrings(FileUtils.java:202)
	at cc.arduino.packages.uploaders.MergeSketchWithBooloader.merge(MergeSketchWithBooloader.java:44)
	at processing.app.debug.Compiler.mergeSketchWithBootloaderIfAppropriate(Compiler.java:1211)
	at processing.app.debug.Compiler.compile(Compiler.java:462)
	at processing.app.debug.Compiler.build(Compiler.java:113)
	at processing.app.Sketch.build(Sketch.java:1109)
	at processing.app.Sketch.build(Sketch.java:1084)
	at processing.app.Editor$BuildHandler.run(Editor.java:1955)
	at java.lang.Thread.run(Thread.java:745)
```

I don't have this problem with 1.6.5r2. Burning the bootloader works fine. I think it's caused by the "MergeSketchWithBooloader" thing only looking in the Arduino install folder for the bootloader. Here's the [boards.txt file](https://github.com/codebendercc/Ariadne-Bootloader/blob/master/hardware/ariadne/avr/boards.txt) from the board I have selected.

ffissore, 2015-06-22:
I see. Sketch merge should be skipped if bootloader file is missing. Because file `ariadne/ariadne_atmega328.hex` is missing, right?

ffissore, 2015-06-22:
Anyway, fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

per1234, 2015-06-22:
No it's not missing, it's located at Arduino15\packages\Ariadne-Bootloader\hardware\avr\1.0.0\bootloaders\ariadne\ariadne_atmega328.hex if I install with Boards Manager or [sketchbook]\hardware\Ariadne-Bootloader\hardware\ariadne\bootloaders\ariadne\ariadne_atmega328.hex
Note that actually the error is for ariadne_atmega2560.hex above but both of the files are present.

ffissore, 2015-06-22:
Ahh indeed. The warning tells me the IDE is looking for that file under the bundled avr core location. That may be another issue: checking

ffissore, 2015-06-22:
Opened #3395

commit 978c8fc6 (2015-06-22): Compiler: don't merge if bootloader file is specified but missing. Print a warning instead. Fixes #3394
closed 2015-06-22 by ffissore