arduino/Arduino #3398: Arduino 1.6.5-r2 installer Java errors

brv64b, 2015-06-22:
Hi
What's with all the Java errors (>10) during install ?
You can ignore them (by hand) and the Arduino IDE will run - but what's wrong ..?
Regards
bs

ffissore, 2015-06-22:
What are those errors? Haven't seen them yet

brv64b, 2015-06-22:
[image: Billede indsat i tekst 1]

[image: Billede indsat i tekst 2]

[image: Billede indsat i tekst 3]

And so on for 10 more screens - all that I Ignored ..
However the IDE works so I ques that it's something local problem with my
pc/setup ..?

ffissore, 2015-06-22:
Sorry, github does not support attachment. Please attach those files using your browser

brv64b, 2015-06-22:
I hope this works ..
[image]
[image]
[image]

PaulStoffregen, 2015-06-22:
Perhaps you have anti-virus or security software which is interfering with the installation?

ffissore, 2015-06-23:
Or perhaps you need to run it as administrator?

brv64b, 2015-06-23:
Nope - that didn't help either ..  :-(
Well - let's leave it there. The IDE works (never the less).
Strange that I should be 'the only one' ..  ;-)

closed 2015-06-23 by ffissore