arduino/Arduino #3399: Arduino Pro 5V schematics missing?

matthijskooijman, 2015-06-22:
On the [Arduino Pro product page](http://www.arduino.cc/en/Main/ArduinoBoardPro), a schematic is listed that looks like the 3.3V version. For the Pro Mini, the schematic is annoted to suggest that the difference between 3.3 and 5 is just a different regulator, but for the Arduino Pro this would mean that the pin labeled "3.3V" actually has 5V and there is no 3.3V available, which seems wrong to me.

Looking at the [sparkfun page](https://www.sparkfun.com/products/10915) it seems they have a newer version of the schematic (v14, Arduino.cc has v12) that labels both power pins (5V and 3.3V on the Uno) as "VCC" and connects them to the 3.3V or 5V regulator output.

Perhaps the schematic and images of the Arduino Pro page should be updated? (on a related note, it seems the Pro Mini page has the latest v14 schematic already, but the images still show an older version)

agdl, 2015-07-17:
@matthijskooijman i pudated the files using the last sparkfun's one. If they are ok for you close this issue please!

agdl, 2015-07-23:
I'm closing this. Reopen if needed

closed 2015-07-23 by agdl

matthijskooijman, 2015-07-23:
Sorry, missed your earlier message. Pro page looks good to me. For completeness, perhaps also update the images on the Pro Mini page? The ones shown now don't have the SJ1 solder jumper yet, sparkfun has the latest version: https://www.sparkfun.com/products/11114

agdl, 2015-07-23:
Done|