arduino/Arduino #3401: String class buffer overflow prevention

RobTillaart, 2015-06-22:
Bug affects these functions

String::String(float value, unsigned char decimalPlaces)
{
	init();
	char buf[33];
	*this = dtostrf(value, (decimalPlaces + 2), decimalPlaces, buf);
}

String::String(double value, unsigned char decimalPlaces)
{
	init();
	char buf[33];
	*this = dtostrf(value, (decimalPlaces + 2), decimalPlaces, buf);
}

if decimalPlaces is greater than the size of buffer the buffer will overflow

Code should limit the decimalPlaces parameter in both functions like this.
Furthermore given the fact that an IEEE754 double has only ~17 significant digits and a float max 9 a buffer size of 20 (double)  resp  12 (float) should be enough instead of 33.

String::String(double value, unsigned char decimalPlaces)
{
	init();
	char buf[33];
        if (decimalPlaces > 30) decimalPlaces = 30;     
	*this = dtostrf(value, (decimalPlaces + 2), decimalPlaces, buf);
}

flicker581, 2016-12-23:
Limit on decimalPlaces will not really work. Second argument to dtostrf is **minimum** width, it does not limit length of the number. Double can be quite large.

sandeepmistry, 2016-12-23:
Hi @RobTillaart @flicker581,

I think pull request #5369 resolves this, could you please take a look and try it out?