arduino/Arduino #3406: SPI Documentation Needs Improvement

PaulStoffregen, 2015-06-23:
This reference page needs to be updated.

https://www.arduino.cc/en/Reference/SPI

This text should be changed:

    To write code for a new SPI device you need to note a few things:

      -  Is data shifted in Most Significant Bit (MSB) or Least Significant Bit (LSB) first?
         This is controlled by the SPI.setBitOrder() function.
      -  Is the data clock idle when high or low? Are samples on the rising or falling
         edge of clock pulses? These modes are controlled by the SPI.setDataMode()
         function.
      -  What speed is the SPI running at? This is controlled by the
         SPI.setClockDivider() function.

I recommend replacing it with this:

    To write code for a new SPI device you need to note a few things:

      -  What is the maximum SPI speed your device can use? This is controlled by the
         first parameter in SPISettings.  If you are using a chip rated at 15 MHz,
         use 15000000.  Arduino will automatically use the best speed that is equal to
         or less than the number you use with SPISettings.
      -  Is data shifted in Most Significant Bit (MSB) or Least Significant Bit (LSB) first?
         This is controlled by second SPISettings parameter, either MSBFIRST or LSBFIRST.
         Most SPI chips use MSB first data order.
      -  Is the data clock idle when high or low? Are samples on the rising or falling
         edge of clock pulses? These modes are controlled by the third parameter in
         SPISettings.

This text should also be replaced:

    The SPI.setDataMode() function lets you set the mode to control clock polarity and phase.

    Every SPI device has a maximum allowed speed for SPI Bus. The SPI.setClockDivider()
    allows you to change the clock speed to make your device working properly (default is 4MHz).

    Once you have your SPI parameters set correctly you just need to figure which registers
    in your device control which functions, and you're good to go. This will be explained
    in the data sheet for your device.

I recommend replacing it with this:

    Once you have your SPI parameters, use SPI.beginTransaction() to begin using the SPI
    port.  The SPI port will be configured with your all of your settings.  The simplest
    and most efficient way to use SPISettings is directly inside SPI.beginTransaction().
    For example:

    SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE0));

    If other libraries use SPI from interrupts, they will be prevented from accessing SPI
    until you call SPI.endTransaction().  Your settings remain in effect for the duration
    of your "transaction".  You should attempt to minimize the time between before you
    call SPI.endTransaction(), for best compatibility if your program is used together
    with other libraries which use SPI.

    With most SPI devices, after SPI.beginTransaction(), you will write the slave select
    pin LOW, call SPI.transfer() any number of times to transfer data, then write the SS
    pin HIGH, and finally call SPI.endTransaction().

These 3 pages should have a notice added.

https://www.arduino.cc/en/Reference/SPISetBitOrder
https://www.arduino.cc/en/Reference/SPISetClockDivider
https://www.arduino.cc/en/Reference/SPISetDataMode

I recommend adding this:

    This function should not be used in new projects.  Use SPISettings with SPI.beginTransaction()
    to configure SPI parameters.

agdl, 2015-06-25:
@PaulStoffregen I added the informations. If they seems good for you please close the issue :)

PaulStoffregen, 2015-06-25:
Thanks.  Looks much better!  :-)

Could you also update this page?  Sorry, I really should have supplied text for it in my original request.  It really needs improvement too.
https://www.arduino.cc/en/Reference/SPISettings

I'd recommend this text:

    Description

    The SPISettings object is used to configure the SPI port for your SPI device.
    All 3 parameters are combined to a single SPISettings object, which is given
    to SPI.beginTransaction().

    When all of your settings are constants, SPISettings should be used directly
    in SPI.beginTransaction().  See the syntax section below.  For constants, this
    syntax results in small and faster code.

    If any of your settings are variables, you may create a SPISettings object
    to hold the 3 settings.  Then you can give the object name to
    SPI.beginTransaction().  Creating a named SPISettings object may be
    more efficient when your settings are not constants, especially if the maximum
    speed is a variable computed or configured, rather than a number you type
    directly into your sketch.

    Syntax

    SPI.beginTransaction(SPISettings(14000000, MSBFIRST, SPI_MODE0))     (Best if all 3 settings are constants)

    SPISettings mySettting(speedMaximum, dataOrder, dataMode)            (Best when any setting is a variable)

    Parameters

    speedMaximum: The maximum speed of communication.  For a SPI chip rated up
                  to 20 MHz, use 20000000.

    dataOrder: MSBFIRST or LSBFIRST

    dataMode: SPI_MODE0, SPI_MODE1, SPI_MODE2, or SPI_MODE3

agdl, 2015-06-26:
@PaulStoffregen Done!

PaulStoffregen, 2015-06-26:
Can you add white space and make these words "(Best if all 3 settings are constants)" and "(Best when any setting is a variable)" italic style?

See "(Arduino Due Only)" on this page for an example.
https://www.arduino.cc/en/Reference/SPISetBitOrder

Those extra words are not part of the actual programming syntax, so separating them from the actual code syntax is important.

agdl, 2015-06-26:
Done!

PaulStoffregen, 2015-06-26:
I see they're italic :)

...but no extra white space like all the other pages.  :(

agdl, 2015-06-26:
i did it in another way that maybe is more clear

PaulStoffregen, 2015-06-26:
looks good  :)

closed 2015-06-26 by PaulStoffregen