arduino/Arduino #3408: Invalid zip file for MATLAB support package

youbz, 2015-06-23:
Error while installing Arduino support package in MATLAB with the following error message

Error installing third party software Arduino software (1.5.6-r2 Beta version): 
Error executing command
"unzip('/Users/_______/Documents/MATLAB/SupportPackages/R2015a/downloads/arduinoio_download/arduino-1.5.6-r2-macosx.zip', '/Users/_______/Documents/MATLAB/SupportPackages/R2015a/arduino-1.5.6-r2')"

Invalid zip file "/Users/_______/Documents/MATLAB/SupportPackages/R2015a/downloads/arduinoio_download/arduino-1.5.6-r2-macosx.zip".

Tried downloading and installing separately and still have the issue. Could not manually extract the zip file either.

ffissore, 2015-06-25:
The zip file unpacks correctly. Please try downloading it again

closed 2015-06-25 by ffissore

youbz, 2015-06-25:
I can't seem to get it to work. I've tried deleting the files and
downloading through Matlab, as well as downloading the package installer
online. I've even tried installing Matlab on windows through bootcamp, but
I get the same error when trying to install the Arduino package.