arduino/Arduino #3411: 'dtostrf' was not declared in this scope - ONE and DUE, but works for UNO, Yun and ATMEGA2560

ad5jo, 2015-06-24:
Hello,
Please will someone add this function for the ONE and DUE boards?
'dtostrf' was not declared in this scope
Thank you,
Dave
void loop() {
  // put your main code here, to run repeatedly:
  // Convert_float_to_String or Double_to_String
  float floatVar = 9.876543;
  int minStringWidthIncDecimalPoint = 16;
  int numVarsAfterDecimal = 8;
  static char charBuf[16];
  String sVal;
  dtostrf(floatVar, minStringWidthIncDecimalPoint, numVarsAfterDecimal, charBuf);
  //dtostrf(floatVar,16, 8, charBuf); 
  sVal = String(charBuf);
}

matthijskooijman, 2015-06-24:
What is the ONE exactly? On the Arduino Uno, the code you pasted works as-is in git master (1.6.5-ish). On the Due, it doesn't, but adding `#include <avr/dtostrf.h>` fixes this.

Also, `dtostrf` is not part of the supported Arduino API. It is offered by avr-libc, but not libc on other architectures. On the Due, there is a dtostrf.h for compatibility, really intended for the `String` class. You can use it as well, but it's not part of the supported API.

The standard way to parse floats using the supported Arduino API is to use `String::parseFloat` (IIRC) method. If you want a portable sketch, use that. Otherwise you will have to use #ifdefs to figure out the current platform and whether `dtostrf` is available and where.

ad5jo, 2015-06-24:
Hi matthijskooijman,
Thank you!
The 'ONE' was a mistake. I meant Zero. Sorry.
This is the URL: http://www.arduino.cc/en/Main/ArduinoBoardZero

Adding  #include <avr/dtostrf.h>  fixes my problem.
I tried to look for  String::parseFloat but I didn't see anything related to Arduino.
I also looked here: http://www.arduino.cc/en/Reference/StringObject
Many Arduino user's have looked for a way to convert String to Float.
Can you prove a URL for  String::parseFloat so I can convert a string to a float in a "Standard" way ?

matthijskooijman, 2015-06-26:
Ah, my bad, it is called [String::toFloat()](https://www.arduino.cc/en/Reference/StringToFloat).

closed 2015-06-26 by matthijskooijman

emanavas, 2019-05-29:
Thanks. works great!

baqwas, 2019-09-29:
What is the "official" solution for the MKR boards, please?