arduino/Arduino #3418: Hosting location of package_index.json seems slow

rogerclarkmelbourne, 2015-06-24:
Guys,

This isn't really a code issue. But I'm noticing that the package_index.json file is very slow to download.

I suggest you move it to GitHub and use their CDN rather than using the Arduino.cc server which doesn't appear to be serving that file very quickly

mastrolinux, 2016-09-16:
Hi @rogerclarkmelbourne we just increased the download servers speed. Please try it again and let me know. We have some geolocation issues and are serving the content in USA only but will let you know if we will be able to improve that.

closed 2016-09-19 by mastrolinux

mgf909, 2016-10-25:
There is something very wrong at the moment downloading anything from downloads.arduino.cc - even the download for the IDE just sits there waiting.

`

--2016-10-25 22:49:52--  (try: 8)  https://downloads.arduino.cc/arduino-1.6.12-windows.zip
Connecting to downloads.arduino.cc (downloads.arduino.cc)|158.69.86.55|:443... failed: Connection timed out.
Retrying.

--2016-10-25 22:52:08--  (try: 9)  https://downloads.arduino.cc/arduino-1.6.12-windows.zip
Connecting to downloads.arduino.cc (downloads.arduino.cc)|158.69.86.55|:443... failed: Connection timed out.
Retrying.

--2016-10-25 22:54:24--  (try:10)  https://downloads.arduino.cc/arduino-1.6.12-windows.zip
Connecting to downloads.arduino.cc (downloads.arduino.cc)|158.69.86.55|:443... connected.
GnuTLS: The TLS connection was non-properly terminated.
Unable to establish SSL connection.
root@hal:~# wget https://downloads.arduino.cc/arduino-1.6.12-windows.zip
--2016-10-25 22:58:31--  https://downloads.arduino.cc/arduino-1.6.12-windows.zip
Resolving downloads.arduino.cc (downloads.arduino.cc)... 158.69.86.55
Connecting to downloads.arduino.cc (downloads.arduino.cc)|158.69.86.55|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 156633336 (149M) [application/zip]
Saving to: ‘arduino-1.6.12-windows.zip’

arduino-1.6.12-windows.zip            0%[                                                                      ]  23.64K  49.1KB/s
`

Times are Sydney AEST