arduino/Arduino #3424: Wi-Fi shield doc fails to mention HDG204

real-or-random, 2015-06-26:
This mentions the old HDG104:
https://www.arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading

It is not clear from the docs if the firmware need/may be updated for the HDG204 as well.
According to an email from the manufacturer, the answer is yes:
http://forum.arduino.cc/index.php?topic=217333.msg1588996#msg1588996

real-or-random, 2015-06-26:
Also, one of the links points to the obsolote arduino/wifishield project.

The correct path is https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares/wifishield

real-or-random, 2015-06-26:
More issues:
 * The new revision of the shield has the AT32UC3A1512 and not the AT32UC3A1256. The difference is in the size of the flash memory. Using the command with "AT32UC3A1256" as parameter still works, but is technically the wrong way and is confusing. I have not tried flashing with "AT32UC3A1512", though. See http://forum.arduino.cc/index.php?topic=189796.0
 * On Windows, you need to install the AVR32 USB driver. This is one source but I guess not the official one: http://www.dfrobot.com/image/data/TEL0064/AVR32%20USB%20Driver.zip (see http://forum.arduino.cc/index.php?topic=161778.0)
 * https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares/wifishield/binary contains also contains files with a suffix _2_1. Are those the newest or the files without suffix? I don't know... 
 * I had to re-attach the shield to the USB port between the two flashing commands.
 * The flip tool needs a 32-bit JRE, which is stated nowhere in the docs nor on the download page of the flip tool. A 64-bit JRE do not work.

agdl, 2015-07-08:
@real-or-random i corrected the page according to your observations. Please take a look and close the issue if it seems good. Yes the _2_1 is the latest firmware

real-or-random, 2015-07-09:
@adgl 

Thanks for your corrections. I think it's still a little misleading
 * Now, the owners of the old module are irritated, because only the HDG204 is mentioned.
 * The same holds for the AT32UC3A1512 vs AT32UC3A1256. The new command probably does not work on the old wifi module (or could hurt it even?)
 * The AVR32 USB driver is not mentioned.
 * You say that the _2_1 is the newest firmware but you did not update the docs. However, I can remember that I tired the _2_1 after opening the issue and it did not work at all, so I had to go back to the files without suffix.
 * Nit: A 64-bit JRE doES not work.

agdl, 2015-07-10:
@real-or-random take a look now. I mention both of them

agdl, 2015-07-23:
I'm closing this, Reopen if needed

closed 2015-07-23 by agdl