arduino/Arduino #3427: Arduino 1.6.5  compiles but hex file NG

EVADK, 2015-06-26:
I have stand alone robotic boards (use 168 or 328) LCD, PWM, etc.
Have been using IDE 1.0.x for years no problem - switched to 1.6.5 r2 and every thing compiles no errors -BUT- looking at the hex files they are corrupt.. uploaded things - goes bonkers But LCD shows fine.. timing from RC input pulses go nuts -----

Tried every board type - 
Switch back to 1.0.5 works fine hex file ok so not my hardware but something screwed up. in my rather large sketch that 1.6.5 does not like .

burned the sketch to an uno - pulled the chip - put it in robot - BAHHHHH.
burned same sketch to same uno with 1.0.5 -- works in robot
Newbie to here so be gentle ( I am over 70 )

Thanks for any insight - ( I know - without any pictures or .ino file )

matthijskooijman, 2015-06-28:
You say the hex file is corrupt, which would make me think it doesn't upload at all, or doesn't do anything at all when uploaded. However, if I understand you correctly, the code is running, it just isn't working exactly like you expect?

I suspect that the problem is a bug in your code. You likely have an error in your code that causes "undefined behaviour", such as reading outside of a buffer. Undefined means that the exact behaviour will depend on (optimization) choices made by the compiler, the order in which things happen, etc. It's likely that with 1.0.5, it just happens to work because of how the compiler did some things. In 1.6.x, there is a newer compiler version, which will likely optimize things differently, exposing your bugs.

Since this issue tracker is not intended to find bugs in people's own sketches, I suggest you post your code to the Arduino forum, hopefully someone there can look over your code and spot the problem(s). Alternatively, you can just debug your code to find the problem (using 1.6.x of course), by e.g. adding debug print output to figure out at what point the code stops doing what you expected it to do.

I'm closing this issue. If I misunderstood and you think there is a problem with the IDE after all, feel free to reopen.

closed 2015-06-28 by matthijskooijman

EVADK, 2015-06-28:
Thank you for the reply   --  the code loads and runs perfect in all 1.0.x versions.  Same code has been running for years.  
I compared the HEX codes - way off  -- and the 1.6.5 is compiling without errors BUT it has areas skipped. My LCD works perfect,  But things go rather complex on RC input !  
I did not want to waste anyone's time but something is changed enough to do this... The difference in compilers tells me my code must not be as tight as it should be!

It appears the IDE version 1.5 and up change dislikes something in my code.  I am sure it is something in the code .  I have warnings (been using the debug option also)  and went after them one by one but no luck yet.  Pulse_in does not work on one input when compiled and loaded on any new version...  works fine on all the 1.0.x  I coded little traps to send numbers to serial port to see what is not working....  Thought someone may have run into this problem.   I will post anything I find and what I may have done in code that is not to the liking of the new IDE compiler... Thank you again for your time and effort...

Chris--A, 2015-06-29:
The newer compiler (released in 1.5.7) is far less forgiving to logical errors. It is 99.99% likely that you have one in your code. and probably 0% likely you have found a compiler bug.

Check your array bounds.
Read up on C++ sequence points, as valid looking code may not be valid at all.

These are all incorrect due to how C++ deals with sequence points. 
```arduino
func(a,a++);
a[i++] = i;
a[i] = i++;
```

EVADK, 2015-06-29:
Thank you for the advice --- I FOUND THE PROBLEM!  It seems that Pulse_in has changed somehow.  I went back to an original copy made no changes but the time out and low and behold all the code works (have warnings to work on) but my 850 lines of code compiled and worked as it does on 1.0.x ...  I increased the timeout and started to pick up RC (radio control) pulses and smooth them properly.  The pulses control so many algorithms - including shutdowns,PWM, and various arrays.  BUT I need to clean up my code even though it has been reliable ---- Thanks   I am on 1.6.5 and noticed 1.6.6 milestone - YES!

matthijskooijman, 2015-06-30:
Awesome! I suspect that the pulseIn function might have become more accurate, so that it previously happened to work because it didn't behave exactly as it supposed to (I remember some changes to pulseIn recently).