arduino/Arduino #3438: Request to Add the SmartEverything Libraries in the Library Manager

smkkHw, 2015-06-29:
please add the following libraries:

request for SmartEverything LE51-868 SIGFOX module
repo = https://github.com/ameltech/sme-le51-868-library/
tag = https://github.com/ameltech/sme-le51-868-library/tree/'V1.0.0'

request for SmartEverything NT3H1X01 Near Field Communication chip
repo = https://github.com/ameltech/sme-nt3h1x01-library/
tag = https://github.com/ameltech/sme-nt3h1x01-library/tree/'V1.0.0'

request for SmartEverything vl6180x proximity sensor
repo = https://github.com/ameltech/sme-vl6180x-library
tag = https://github.com/ameltech/sme-vl6180x-library/tree/'V1.0.0'

request for SmartEverything library to manage the Humidity Sensor
repo = https://github.com/ameltech/sme-hts221-library
tag = https://github.com/ameltech/sme-hts221-library/tree/'V1.0.0'

request for  SmartEverything library to manage the GPS component
repo = https://github.com/ameltech/sme-se868-a-library
tag = https://github.com/ameltech/sme-se868-a-library/tree/'V1.0.0'

request for SmartEverything library to manage the 9Axis component
repo = https://github.com/ameltech/sme-lsm9ds1-library
tag = https://github.com/ameltech/sme-lsm9ds1-library/tree/'V1.0.0'

request for  SmartEverything LPS25H pressure sensor.
repo = https://github.com/ameltech/sme-lps25h-library
tag = https://github.com/ameltech/sme-lps25h-library/tree/'V1.0.0'

Thank, Mik

ffissore, 2015-06-30:
Added. Thanks!

closed 2015-06-30 by ffissore

smkkHw, 2015-06-30:
thank you for your prompt response