arduino/Arduino #3439: Starting IDE when IDE is already open with a sketch opens another copy of that sketch

per1234, 2015-06-30:
Using Arduino IDE 1.6.6 Hourly Build 2015-06-29 08:41 with Windows 7 64 bit
- Open a saved sketch in the IDE
- Start IDE - There are now two copies of the previously opened sketch open

If multiple sketches are open when the second instance of the IDE is started then it creates duplicate windows of all open sketches.

This is similar behavior to reopening the IDE after File > Quit so it may be related to that code.

ffissore, 2015-06-30:
If I got this right, you have started the IDE twice. If so, although weird in some cases, it is expected and it's the price to pay for allowing to start multiple and different versions of the IDE

per1234, 2015-06-30:
Yes that's correct I started the IDE twice but both the same version not different versions. I would expect that to just have the same effect as File > New but it's no big deal just thought I'd mention it in case you considered it a bug.

ffissore, 2015-06-30:
It's not. We just allow multiple instance of the IDE to be started and don't make then check what the others have done/are doing. Closing as invalid then

per1234, 2018-03-31:
I'm reopening this issue for reconsideration because "weird" behavior when multiple instances of the Arduino IDE running should be considered a bug (even if not a high priority one) since that usage provides the very useful feature of allowing different boards to be selected and multiple Serial Monitors/Serial Plotters to be open (see https://github.com/arduino/Arduino/issues/7337#issuecomment-373924001, https://github.com/arduino/Arduino/issues/6819#issuecomment-336074269, https://github.com/arduino/Arduino/issues/1482#issuecomment-312519165).

per1234, 2024-01-28:
Closing as resolved in Arduino IDE 2.x. Arduino IDE 2.x only allows a single instance to be open at a time. Subsequent invocations of the IDE will simply open the existing instance rather than a duplicate instance.

closed 2024-01-28 by per1234