arduino/Arduino #3440: Pick up changes in .cpp files when compiling

probonopd, 2015-06-30:
I have a sketch which includes a cpp file in the same directory as the .ino file. When I make a change to the .cpp file, save, and compile, then the change in the .cpp file is not picked up. I have to restart the Arduino IDE each time to have it recognize the change in the .cpp file.

It would be nice if the restart wasn't necessary. (This might apply to other file types different from .cpp as well).

matthijskooijman, 2015-06-30:
Hm, weird. In any case, there have been some recent changes in how these files are handled. Could you try running the hourly build and see if the problem still exists there?

probonopd, 2015-06-30:
I am running 
[image]

matthijskooijman, 2015-06-30:
Hm, I was talking about #3435, which was merged only four hours ago. You might want to upgrade to the latest hourly, just in case.

@ffissore, might be good to also list the git revision of the latest hourly on the site and in the IDE itself?

ffissore, 2015-06-30:
Ouch, confirmed. Working on it

ffissore, 2015-06-30:
Fixed. Sorry about that. Available since next hourly http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-30 by ffissore

probonopd, 2015-06-30:
Thanks ffissore. I can confirm it is fixed.

commit 9a01e306 (2015-06-30): Compiler: don't filter out existing sketch files, overwrite them. Fixes #3440