arduino/Arduino #3441: Save expand/collapse code folding!

MrKp91, 2015-06-30:
I've recently found under preference the ability to have code folding which is very helpful but every time I switch tabs everything is expanded.

This is very annoying specially when you have  collapse specific parts of your code and go check something on another tab just to have it all expanded.

ALSO, it would be nice if there's a shortcut to collapse/expand everything.

Thanks!

ffissore, 2015-07-01:
That will take a while to get solved. The root cause is that we use the same editor for every tab, replacing its content every time your switch one. This resets its previous settings, like folded code blocks.
This is a legacy behaviour, which forced us event to code a customized version of the new editor (RSyntaxTextArea)
Solution is to stop recycling the editor and create new ones for each tab. Once done, this issue should autosolve

lmihalkovic, 2015-12-15:
This is another example of a very low hanging fruits in this project (just added the feature to my branch despiteand it works fine), and blaming the editor-sharing is just a bad excuse for not implementing the feature.

There are fundamentally not technical justification for the customized version of RSTxtArea, and this one owes more to poorly structured code than to real requirements. That is one of the issues pleaguing this industry: "It can't be done" used liberally in place of "I don't know how to do it".

lmihalkovic, 2015-12-16:
just in case... this is to prove that it is indeed extremely simple to implement:

```
  @Override
  protected void configureFolding(SketchFileType type) {
    textarea.setCodeFoldingEnabled(true);
    textarea.getFoldManager().addPropertyChangeListener(new PropertyChangeListener() {      
      @Override
      public void propertyChange(PropertyChangeEvent evt) {
        if(evt.getPropertyName() == FoldManager.PROPERTY_FOLDS_UPDATED) {
          @SuppressWarnings("unchecked")
          List<Fold> list = (List<Fold>)evt.getNewValue();
          state().foldings = list;
        }        
      }
    });
  }  

  @Override
  public void restoreState() {
    List<Fold> list = state().foldings;
    textarea.getFoldManager().setFolds(list);
  }  
```

lmihalkovic, 2015-12-25:
@cmaglie @matthijskooijman   should you have the availability within the near future, it would likely be very appreciated if you'd look into this one. The code does work *as-is* in my fork, but will need some adapting for your codebase. I gave a short description of what it boils down to [here](https://forum.arduino.cc/index.php?topic=366459.0), there is really nothing more to it. (oops.. and yes, surprisingly it works despite the very stupid *==* which I will fix).

matthijskooijman, 2016-01-02:
@lmihalkovic your fix seems good to me, with the current code. However, in #4363 I opted for a different fix: Keep a different RSyntaxTexArea for each tab. Even though RSyntaxTextArea seems to support switching between multiple documents ([and some improvements were made in this area](https://github.com/bobbylight/RSyntaxTextArea/issues/84#issuecomment-66397658)), but in the end the author still somewhat [recommends just using a separate instance for each tab](https://github.com/bobbylight/RSyntaxTextArea/issues/110#issuecomment-87317168). I recalled that there was a more explicit recommendation than the one I mentioned above, but I could not find it just now.

lmihalkovic, 2016-01-06:
@matthijskooijman  the code is and always was irrelevant. I only shared it to prove how utterly wrong the initial evaluation of the severity of the problem/complication of the solution were. Fixing this issue is and always was a trivial matter which should have only taken 20 minutes the days this ticket was created.

This was my point... i have watched this repo for 2 months now, and I am truly surprised by the number of issues where similarly wrong assessments exist to justify not addressing issues that should have taken, in many instances, at best a couple days of a programmer's time. 

And IMHO #4363 goes one step further by destroying useful flexible constructs only to replace them with an equally arbitrary, but IMHO less practical, choice of entanglement. But i fully appreciate that it will take you a number of days/months/years (looking at a number of things i saw, i would wager that you have not written much java code yet) before we can even have a meaningful debate on the topic.

Jim4Beam, 2016-06-20:
Is there a work arround or solution for the  ode folding issue yet? Sorry I am new to programmig.

matthijskooijman, 2016-06-20:
@Jim4Beam, AFAIU the PR that fixes this is scheduled to be fixed shortly after the next Arduino release.

closed 2016-08-26 by facchinm

TehseenHasan, 2020-06-12:
any solution for code folding as this issue stated in the title?