arduino/Arduino #3445: IDE on windows doesn't loads if username contains non-ASCII characteres

vfrico, 2015-07-01:
Hello
I've tried to load Arduino IDE under windows (8.1 and 10) and this never starts. After a very short time of bootscreen, it suddenly stops. Running arduino_debug.exe under administrator made me realize that it tried to access to home folder of an inexistent user. I mean: my user on Windows is "Víctor", and the folder is on c:\Users\Víctor. But Arduino tried to look for (exactly): c:\Users\V!ctor. Obviously, the folder didn't exist and it breaks always i try to run IDE

The ide should be able to deal with name folders with non-ascii characteres. Otherwise some users can experience big troubles

Chris--A, 2015-07-02:
This looks like it has been fixed: https://github.com/arduino/Arduino/issues/3360

You can re-download a version with the latest updates in the 'latest hourly'section: https://www.arduino.cc/en/Main/Software#hourly

ffissore, 2015-07-02:
Confirm. Positive feedback about the fix has been reported as well. See https://github.com/arduino/Arduino/issues/3360#issuecomment-114548364

closed 2015-07-02 by ffissore