arduino/Arduino #3446: "efuse" memory type not defined for part "atmega8" Error while burning bootloader. In arduino/hardware/arduino/avr/boards      for atmega8 is good declared low fuses and high fuses, and is not present declaration for extended fuses like example atmega168, so is good, but the function who realy do the fuses try to fuse and extended fuses for atmega8, and atmega8 has no extended fuses, so gives that error.  I know that atmega8 is old" but i want to use it with bootloader ... Please correct the function who realy make the fuses for atmega8 for not try to fuse extended fuses. Or tell me where is it , to try to correct myself

andrei777777777, 2015-07-02:
"Efuse" memory type not defined for part "atmega8"
Error while burning bootloader.
In arduino/hardware/arduino/avr/boards      for atmega8 is good declared low fuses and high fuses, and is not present declaration for extended fuses like example atmega168, so is good, but the function who realy do the fuses try to fuse and extended fuses for atmega8, and atmega8 has no extended fuses, so gives that error.
 I know that atmega8 is old" but i want to use it with bootloader ...
Please correct the function who realy make the fuses for atmega8 for not try to fuse extended fuses.
Or tell me where is it , to try to correct myself

rigelinorion, 2015-07-02:
Try this https://github.com/arduino/Arduino/issues/2541

andrei777777777, 2015-07-02:
Yes!!   I deleted the line with efuse.

It worked thank you, but now will not work the others wich need efuse.

Ofcourse i made a copy of original platform, and when i will need it for others microcontrolers i will replace it, but you must solve this problem for arduino.. You know it's wery unprofetional like this. 

Trimis din Yahoo Mail pe Android

De la:"rigelinorion" <notifications@github.com>
Data:Jo 2 iul. 2015 la 13:57
Subiect:Re: [Arduino] "efuse" memory type not defined for part "atmega8" (#3446)

Try this #2541

rigelinorion, 2015-07-03:
I do not think you understand this way of working. The solution is not removing EFUS of the line only, but for the creation of additional tools that you use for the mC that this EFUS do not use. You do not need to swap any files.

per1234, 2016-09-25:
@cmaglie this is a duplicate of https://github.com/arduino/Arduino/issues/2075 in which is described IMO the best way to solve this issue at https://github.com/arduino/Arduino/issues/2075#issuecomment-238031689.

aakrutii, 2016-10-27:
i have solved this problem check this link below    @andrei777777777 @rigelinorion @per1234 @cmaglie @JackDanger 
https://www.youtube.com/watch?v=2MMYnIFaXv4

you will get the solution here

per1234, 2017-07-01:
Closed by:
https://github.com/arduino/avrdude-build-script/commit/0bfe2c433ffe9b0371c5ac1adf5bf9ecf812cf12
and
https://github.com/arduino/Arduino/commit/b1f349244b85bd8c39c753dc0e267a1e7a5f01c6

closed 2017-07-01 by per1234