arduino/Arduino #3448: Unsaved changes to non-.ino files are ignored

matthijskooijman, 2015-07-02:
#3435 changed the way additional files in the sketch directory (e.g. everything but .ino or .pde) are handled, to support files in subdirectories too. AFAICS, this means that any unsaved changes in those additional files will be lost, since the files are compiled from their on-disk versions.

To properly fix this, I think the files in subdirectories need to be added to the list of sketch code files (which should also cause them to be opened in the IDE, which is a nice bonus) and the previous behaviour (without the recursive copy) should be restored.

Note that I haven't actually tested any of this, this is just based on my understanding of the code.

ffissore, 2015-07-03:
Well, actually the problem doesn't exist. It would exist if those files in subfolders were editable from the IDE, but they are not, thus of course you need to save them with your external editor if you want to see those changes applied.

This issue should then turn into "support editing of source files stored in subfolders", which we won't. What would look like a simple feature to add it's likely to turn into a whole file manager. How would you manage those files? Should empty folders be removed? Can I just create a folder? How to move files between folders? "With tool X I can just drag and drop files, why can't I do that with the IDE!?!?" Can I rename a folder?

The only way of making it hopefully easy to use is to have a file manager slide panel, which we won't add, at least for now

matthijskooijman, 2015-07-03:
>  It would exist if those files in subfolders were editable from the IDE, but they are not, thus of course you need to save them with your external editor if you want to see those changes applied.

Ah, I see the copy thing only applies to subfolders now, I thought you had left it for files directly in the sketch folder too.

Regarding all those questions you ask - can we even do things like that for files currently? Can I remove a file using the IDE? Can I rename a file? Oh wait, we can, I just found the tab menu on the right :-) And indeed, things will quickly get complicated, I guess.

So I guess the current state of things is actually good enough - you can put things in subdirectories when using an external editor, which will be properly compiled, but the editor does not support it.

closed 2015-07-03 by matthijskooijman