arduino/Arduino #3450: Add #pragma message to header file  - suggestion

VaclavSal, 2015-07-02:
Just downloaded latest nightly 1.6.6. and need to modify print.h.
I have several older IDE's floating around ( copies for safety reasons ) and been looking  for which print.h is being included in my application. 
So far unsuccessfully. 
( OK, I can include full path to bypass OS search for the file. ) 

With so many recent changes It would be nice if header file preporcessor directives look something like this:  

#ifndef HEADER_H
#define HEADER_H
#pragma message ("Induding header.h version x" ) 
#endif 

I use this in my local files and include a #ifdef PRAGMA  directive to avoid unnecessary clutter in compiler output.

ffissore, 2015-07-09:
AFAIK, #pragma works fine. Just added it to the initial empty sketch and verbose compilation reported 
```
sketch_jul09a.ino:3:48: note: #pragma message: Induding header.h version x
```
Is that your issue?

VaclavSal, 2015-07-09:
No, that is not the issue. I know #pragma works. 

But if everybody added this to file they put into public domain 
#pragma message (__FILE__) 
it would actually identify WHICH file is being used by the IDE.

ffissore, 2015-07-10:
Sorry, I don't understand what your asking for. Are you asking everybody to use #pragma your way? Are you asking us to do that?

Chris--A, 2015-07-10:
@VaclavSal 
Hmm, the output window should be left for warnings and errors. With verbose switched on there is enough output already.

The IDE does add a diagnostic already when using a library that has multiple versions (or the same name as another).

VaclavSal, 2015-07-10:
Sorry, I don't understand what you're asking for. Are you asking everybody to use #pragma your way? Are you asking us to do that?

Yes.

Almost every file published by Arduino volunteer developers include big "...this is open source , you should  have received copy of blah blah " 

Almost NONE of them contain ANY description what the file does etc. Very few contains code comments...
As good as recent developments regarding libraries are , they still fall short. 

I agree it is harder to extract warnings from errors, but IMHO more info during development is better than less.

I really believe that getting Arduino from  "blink LED" attitude amateur developers should look around
and LEARN from professionals - take  a look at ATmel code, for example.

But this was just a suggestion.

Chris--A, 2015-07-10:
The licence info is important. Without it you cannot assume that you can freely use the source in your application.

ffissore, 2015-07-10:
If you want more comments and documentation in code, feel free to contribute it. As per pragma, we are not going to pollute the already verbose output that you get enabling verbose output in File > Preferences

closed 2015-07-10 by ffissore

PaulStoffregen, 2015-07-10:
@VaclavSal - Have you ever submitted a pull request?  Even one?  Maybe this, just adding some comments, could be your perfect opportunity to actually contribute in a more meaningful way than lots of rhetoric in comments?!