arduino/Arduino #3451: Serial Plotter examples

pierrotm777, 2015-07-02:
I see a new 'Serial Plotter' feature that arrive with the future IDE 1.6.6 .

Can you give us some examples for it use ?

Thanks

ffissore, 2015-07-02:
There is an example sketch at https://github.com/arduino/Arduino/pull/2177#issuecomment-48763705
It would be nice if you could provide an example good enough to be included in the standard set

pierrotm777, 2015-07-02:
It's fantastic new feature .
I cool option would be to add several plots possibility in same time with different colours :-)

ffissore, 2015-07-02:
/cc @henningpohl in case he wants to jump in

henningpohl, 2015-07-02:
I added a basic example here: henningpohl/Arduino@0f395d88caeb8667e58ac0ca95c917ef53a272e1

It is a bit odd though, as the example is not really showing off capabilities of the Arduino, but instead of the IDE. Would you want a PR for this anyway?

As for extension options: certainly multiple graphs, each with a different color, would be a clear way for this to move. There's a bit of a trade-off in that this could get overly complicated if not kept in check. For example, somebody might want to add spectrograms, logic analysis, or signal smoothing. I personally don't think that's what this should become. But hey, at the end that's really up to the people using this.

ffissore, 2015-07-03:
Thank you @henningpohl, we are reviewing it

ffissore, 2015-07-06:
@henningpohl instead of a new example, we prefer enriching AnalogReadSerial description. See d9b3a86e84b8335d7e48c340b20ec70a0ca3a921

closed 2015-07-06 by ffissore

ffissore, 2015-07-07:
And ReadAnalogVoltage. See 974fef1000ba72edd3c305c6016a3e6da5697770

q2dg, 2015-09-06:
Maybe it should be noticed in these comments that Serial Plotter only works with Serial.println() -because it uses \n as value delimiter, does it?- but not with Serial.print() or Serial.write().

q2dg, 2015-09-06:
@henningpohl The idea of showing multiple graphs...is finally discarded?

henningpohl, 2015-09-29:
@q2dg not discarded, but I won't have time for this before mid-October

q2dg, 2015-09-29:
Ok, thanks a lot. Casually yesterday I wrote a comment in https://github.com/arduino/Arduino/pull/2177 asking for the same...I don't wanna be onerous, sorry

q2dg, 2015-10-26:
@henningpohl Thanks a lot for that pull request!! Let's see if we are lucky and it gets merged...
If it were...how would be a code sending two series of values? Is there some identifier inside the message written by Serial.println()?? Thanks!!

henningpohl, 2015-10-26:
You just need to separate individual values by commas or spaces. So you could, e.g., send one update from the Arduino as: "3.4, 5.4, 1.0\n".

q2dg, 2015-10-26:
Ahhh, terrific!! Thanks!

jonathanrjpereira, 2016-04-01:
Here's a link to a tutorial that I made on Instructables about the Arduino Serial Plotter. It has a detailed explanation & example code too.

http://m.instructables.com/id/Ultimate-Guide-to-Adruino-Serial-Plotter/

commit d9b3a86e (2015-07-06): Serial plotter: added comment to AnalogReadSerial. Fixes #3451