arduino/Arduino #3453: The USB HID core libraries enable by default the USD HID Keyboard emulation

joseangeljimenez, 2015-07-02:
The USB HID core libraries enable by default the USD HID Keyboard emulation. In fact, they enable by default both the emulation of the Keyboard and Mouse emulation.

I have prepared a very simple fix/patch which allows to disable the USB HID Keyboard emulation by uncommenting a #define KEYBOARD_ENABLED in HID.cpp.

I am looking into this so as to further simplify enabling/disabling the Mouse & Keyboard USB HID emulation.

matthijskooijman, 2015-07-02:
Have a look at #3304, I think it pretty much solves this exact same thing.

facchinm, 2015-07-03:
@joseangeljimenez , if you could test PR #3304 and report your impressions it could help a lot its inclusion (and also the solution for your problem :wink: )

closed 2015-07-03 by facchinm